JoelBender/bacpypes #263: ReadProperty on multiState for property stateText fails on 0.17.5 and lower.

amrx101, 2019-03-27:
Environment

1. Python 2.7
2. BACpypes 0.17.5
3. [BACnet Device](https://www.kmccontrols.com/component/kmcproducts/?view=product&alias=bac-120063c)

The read on `multiStateValue` for property `stateText` fails.

`read 192.168.130.108 multiStateValue:1 stateText` gives the following error

`ERROR:__main__.ReadPropertyAnyConsoleCmd:exception: DecodingError('too many cast components',)
Traceback (most recent call last):
  File "ReadPropertyAny.py", line 95, in do_read
    value = apdu.propertyValue.cast_out(datatype)
  File "/Users/modiji/.virtualenvs/all/lib/python2.7/site-packages/bacpypes/constructeddata.py", line 1343, in cast_out
    if len(self.tagList) == 0:`

The culprit seems to be the check on the length of tagList [here](https://github.com/JoelBender/bacpypes/blob/master/py27/bacpypes/constructeddata.py#L1341) as for `ReadProperty` this expects a tag of single item ( I believe the retrieved `datatype` for `objectIdentifier` and `propertyIdentifier` is responsible for that (may be wrong here).

A temporary fix that I did was

 
    if len(self.tagList) > 1:
           values = []
           helper = klass(self.tagList[0])
           for val in self.tagList:
                  helper = klass(val)
                  values.append(helper.value)
           return values

After this change we have the following 

  > read 192.168.130.108 multiStateValue:1 stateText
['NOT CONFIGURED', 'AIR HANDLER', 'FAN COIL'] (<class 'bacpypes.primitivedata.CharacterString'>)

This seems to be a bug and needs to be fixed. I dont thing that my change is correct as there is a mismatch between the `value` and its `datatype`. @JoelBender would love your input on this. Its giving me hibby jibby on production.

Thanks.

JoelBender, 2019-03-27:
Could you please run this again and jam in a `apdu.debug_contents()` call when the response comes back, I need to see the contents of the tag list.  It _should_ be an opening context tag 3, then a sequence of CharacterString application encoded values.  The code that you put in shouldn't have to be there because ArrayOf takes care of it.  I'll try and reproduce it.

amrx101, 2019-03-28:
@JoelBender , here is the result of `apdu.debug_contents()`

    pduSource = <Address 192.168.130.108>
    pduExpectingReply = False
    pduNetworkPriority = 0
    apduType = 3
    apduSeg = False
    apduMor = False
    apduService = 12
    apduInvokeID = 1
    objectIdentifier = ('multiStateValue', 1)
    propertyIdentifier = 'stateText'
    propertyValue
            <bacpypes.primitivedata.Tag(characterString) instance at 0x10e17de10>
                tagClass = 0 application
                tagNumber = 7 characterString
                tagLVT = 15
                tagData = '00.4e.4f.54.20.43.4f.4e.46.49.47.55.52.45.44'
            <bacpypes.primitivedata.Tag(characterString) instance at 0x10e17de90>
                tagClass = 0 application
                tagNumber = 7 characterString
                tagLVT = 12
                tagData = '00.41.49.52.20.48.41.4e.44.4c.45.52'
            <bacpypes.primitivedata.Tag(characterString) instance at 0x10e17de50>
                tagClass = 0 application
                tagNumber = 7 characterString
                tagLVT = 9
                tagData = '00.46.41.4e.20.43.4f.49.4c'
    pduData = x''

JoelBender, 2019-03-28:
The decoding is working correctly (see code) so there's something else going on.  There is a `samples/MultiStateValueObject.py` sample application that has one of these things, run that as your server and hit it from the client and see if that works.
```
from bacpypes.primitivedata import CharacterString, Tag, TagList
from bacpypes.constructeddata import ArrayOf

ArrayOfCharacterString = ArrayOf(CharacterString)

tag_list = TagList()
for some_string in ('what', 'is', 'going', 'on'):
    cstr = CharacterString(some_string)
    t = Tag()
    cstr.encode(t)
    tag_list.append(t)

tag_list.debug_contents()

aocs = ArrayOfCharacterString()
aocs.decode(tag_list)

for i in range(5):
    print(aocs[i])
```
The datatype for that property should be `<class 'bacpypes.constructeddata.ArrayOfCharacterString'>` so you might want to check that in the object class:
```
>>> for x in MultiStateValueObject.properties:
...     if x.identifier == 'stateText':
...         print(x.datatype)
... 
<class 'bacpypes.constructeddata.ArrayOfCharacterString'>
```

amrx101, 2019-03-28:
@JoelBender , by Client code you mean the existing `ReadPropertyAny` or incorporate the above code in `ReadPropertyAny`. Bit confused here :(.

Thanks

JoelBender, 2019-03-29:
You've got four pieces of software:
- (A) the server you are getting data from
- (B) a server running `MultiStateValueObject.py` sample application
- (C) your code that breaks
- (D) the `ReadProperty.py` sample code from the library

The bug is (C) talking to (A), and (D) talking to (B) works.  So if (C) talking to (B) also breaks, it might be (C).  If you test (D) talking to (A) and it breaks then it is probably a library bug, but might be a quirk in (A).  If (D) to (A) succeeds but (C) to (B) breaks then the problem is in your code.

amrx101, 2019-04-02:
@JoelBender  its (D) to (A) that breaks. The original error message is from (D) more precisely `ReadPropertyAny.py` in the `samples` directory.

JoelBender, 2019-04-02:
Ha!  I completely missed that the error message `ERROR:__main__.ReadPropertyAnyConsoleCmd` should have told me it wasn't `ReadProperty.py`.  The sample applications have been updated (see commit messages above).  The "any" version doesn't try to know the datatype from the property identifier so that it will work with vendor properties or ones that BACpypes doesn't know about yet, so the decoding is different than other applications.  Sorry for the confusion!

closed 2019-04-11 by JoelBender