JoelBender/bacpypes #264: LocalTrendLogObject

JoelBender, 2019-04-19:
Create a **LocalTrendLogObject** with all the right stuff for maintaining the log which will need an additional API for interpreting the Range choices as well as something like `log_status()`, `log_value()`, `log_failure()`, and `log_time_change()`.  Continuation of #207.  This might include a  **LocalEventLogObject** as well.

kazemiprince, 2020-04-10:
any news from this object?
is it implemented or not yet?
I wish to have a such object in BIPSimpleApplication, but I could not find any

JoelBender, 2020-04-14:
Yikes!  Almost a year!  No, nothing yet.  :-(

ChristianTremblay, 2023-02-15:
`cast_in` nightmare.... thought it would be an easy ride

JoelBender, 2023-02-15:
Quite a bit of a pain.  Are you dealing with non-primitive data types in the `LogRecordLogDatum`?

ChristianTremblay, 2023-02-15:
A realValue... i just don't get it. 

The internal data type is a ListOf of a SequenceOfAny depending on how I generate logBuffer internally. 

I end up with some bytes error saying that bytes value should be between 0-256 ... or an invalid constructor. 

This error occurs when I try to readRange the logBuffer from another BAC0 instance. As this is under the doReadRange thing.

ChristianTremblay, 2023-02-16:
Note to myself, remember bacpypes deal with dates as year minus 1900... been hard to tackle down. There should be a "StupidProof" mechanism in the Date class that subtract 1900 if it sees a 2000 date....

This explained the bytes error... and allowed me to fill the itemData of the response
```python
    resp = ReadRangeACK(context=apdu)
    resp.objectIdentifier = objId
    resp.propertyIdentifier = apdu.propertyIdentifier
    resp.propertyArrayIndex = apdu.propertyArrayIndex

    resp.resultFlags = [1, 1, 0]
    resp.itemCount = len(value)

    # save the result in the item data
    item_data = SequenceOfAny()
    item_data.cast_in(value)
    resp.itemData = item_data
    self._log.debug("    - itemData : %r", resp.itemData)
    self._log.debug("    - resp: %r", resp)
    self.response(resp)
``` 

Made me refactor BAC0 to treat logDeviceObjectProperties as otional... 

Each LogRecord is built this way
```python
LogRecord(
            timestamp=_dt,
            logDatum=LogRecordLogDatum(realValue=Real(val)),
            statusFlags=StatusFlags([0, 0, 0, 0]),
        )
```

Now I need to figure a good way to build some kind of database in memory from the reading I get...to then build the logRecord.... I actually done something weak and I get duplicates.

Hope those notes will help others