JoelBender/bacpypes #267: Pytest fails to run

ChristianTremblay, 2019-04-25:
```
$ pytest tests/test_primitive_data/test_character_string.py
ImportError while loading conftest 'C:\0Programmes\Github\bacpypes\tests\conftest.py'.
tests\__init__.py:21: in <module>
    from . import test_utilities
tests\test_utilities\__init__.py:10: in <module>
    from . import test_state_machine
E     File "C:\0Programmes\Github\bacpypes\tests\test_utilities\test_state_machine.py", line 31
E       '{}={}'.format(k, v) for k,v in self.__dict__.items(),
E       ^
E   SyntaxError: Generator expression must be parenthesized

```

JoelBender, 2019-04-26:
This seems to be a pytest thing:
```bash
#!/bin/bash

for ver in 2.7 3.5 3.6 3.7; do
    python$ver -c 'print(", ".join("{}={}".format(k,v) for k,v in {"x":1, "y":2}.items()))'
done
```

JoelBender, 2019-05-24:
Merged

closed 2019-05-24 by JoelBender