JoelBender/bacpypes #270: Is there any sample on how to use the TimeSynchronisationRequest?

mosterme1003, 2019-05-03:
Hello everyone! I'm looking for a way to use the TimeSynchronisationRequest and I don't get how to use it by looking at the service's code. I would really appreciate a sample on how to overwrite/synchronise a device's DateTime. Thank you in advance.

JoelBender, 2019-05-04:
Assuming you want to send a request to another device using your current date and time, try cloning the WhoIsIAm.py sample application and add a `do_tsr()` function.  The request PDU would be something like this:
```python
from bacpypes.pdu import Address
from bacpypes.primitivedata import Date, Time
from bacpypes.basetypes import DateTime
from bacpypes.apdu import TimeSynchronizationRequest

tsr = TimeSynchronizationRequest(
    destination=Address("192.168.0.23"),
    time=DateTime(date=Date().now().value, time=Time().now().value),
)
```

JoelBender, 2019-05-04:
...and to follow up, if you are receiving one of these from some other "time master" then you are providing a `do_TimeSynchronizationRequest()` function in your server code with a bit of glue to put the date and time together:
```python
from time import gmtime

when = gmtime(float(Date(apdu.time.date)) + float(Time(apdu.time.time)))

time.struct_time(tm_year=2019, tm_mon=5, tm_mday=4, tm_hour=3, tm_min=10, tm_sec=59, tm_wday=5, tm_yday=124, tm_isdst=0)
```
Setting your local time to this value is platform dependent.

mosterme1003, 2019-05-06:
Thank you so far! I added the do_tsr as follows and it works by calling from the console.

def do_tsr(self, args):

        date= (119,10,10,4)
        time= (22,22,22,22)
        dateTime = DateTime(date=date, time=time)
        tsr = TimeSynchronizationRequest(
            destination=Address("192.168.0.3"),
            time=dateTime,
        )

        this_application.request(tsr)

Now I want to change the device's date and time by calling the function from another script. I copy-pasted the do_tsr to the WhoIsIAmApplication class, commented out the console part in main() and called the tsr method. The tsr method gets called but nothing happens. Would you help me to find out whats wrong?

class WhoIsIAmApplication(BIPSimpleApplication):

    def __init__(self, *args):   ...

    def request(self, apdu):   ...

    def confirmation(self, apdu):   ...

    def indication(self, apdu):   ...
     
    def timeSynchronisation(self):
   
        date = (119, 10, 10, 4)
        time = (22, 22, 22, 22)
        dateTime = DateTime(date=date, time=time)

        tsr = TimeSynchronizationRequest(
            destination=Address("192.168.0.3"),
            time=dateTime,
        )
        self.request(tsr)

def main():
    
    global this_device, this_application

    # parse the command line arguments
    args = ConfigArgumentParser(description=__doc__).parse_args()

    # make a simple application
    this_application = WhoIsIAmApplication(this_device, args.ini.address,)

    # calling the tsr function
    this_application.timeSynchronisation()

JoelBender, 2019-05-07:
Your outgoing request is queued up until the `run()` function is called (in the `bacpypes.core` module), all of the actual sending and receiving socket activity happens during `asyncore.loop()`, which is called repeatedly until it is told to stop (by calling `stop()`).

`select()`

mosterme1003, 2019-05-16:
Thank you! It works so far. With calling stop() the script remains in the loop for just one run through until it exits and the device's localTime is overwritten. Using PyCharm it works perfectly fine running the TimeSync multiple times in a row. But for a project I'm calling my TimeSync python file in a Matlab script. First time it works, after that I receive the same error every time from python.

> Error using asyncore>bind (line 329)
        Python Error: OSError: [WinError 10048] Only one usage of each socket address 
        (protocol/network address/port) is normally permitted

> Error in udp>__init__ (line 154)
        self.bind(address)

> Error in bvllservice>__init__ (line 96)
        self.directPort = UDPDirector(self.addrTuple)

> Error in app>__init__ (line 514)
        self.mux = UDPMultiplexer(self.localAddress)

> Error in TimeSynchronisationRequest>__init__ (line 51)
        BIPSimpleApplication.__init__(self, *args)

> Error in TimeSynchronisationRequest>main (line 218)
        this_device, ('', 47808),

Now how can I 'disconnect' and 'release' the socket address so I can successfully call the script again send another TimeSyncRequest? Thank you for your time!

closed 2019-07-29 by mosterme1003