JoelBender/bacpypes #271: Threading issues

mcmayer, 2019-05-08:
I'm having some threading issues with `samples/ThreadedReadProperty.py`. When the number of devices (and hence threads) gets large enough I start seeing 'stuck threads'. With that I mean a thread that just seems to stop, without error or any other signs of trouble.

With some debugging (lots of prints, really) I narrowed it down to line 92:
```
iocb.wait()
```
Threads randomly get 'stuck' at that line.

When I create a global `Lock` for all calls into `this_application` the stuck threads are gone, but so is the parallelism.
Can anyone offer any insights? That would be greatly appreciated because I'm totally unsure what to do now.

I'm using the `py34` library.

pshank, 2019-05-14:
I ran into the same exact problem today, where my main thread stalls out at ```iocb.wait()``` rather unexpectedly. I had a heck of a time trying to find the culprit, but so far have had no luck. It looks like I was able to work around the issue by adding the ```sleep``` call per your suggestion, so in my case, I think you are correct that there was simply too much traffic which was causing a request to be thrown out. 
One (possibly dumb) question for you in regards to another of your suggestions.... is there a good way to set the core's logging to ```debug``` if I am not using a command line to kick off the process? I currently have everything integrated into my own custom application which runs on it's own and is being watched by supervisor.

JoelBender, 2019-05-15:
I'm glad the sleep thing worked.  As for debugging, that's easier.  Import the console log handler:
```python
>>> from bacpypes.consolelogging import ConsoleLogHandler
>>> import bacpypes.core
```
And "attach" it to a module, class, or function:
```python
>>> snork = ConsoleLogHandler("bacpypes.core")
```
One of the side effects of attaching the handler is that it turns on the `_debug` flag in the module.  This flag is a fast way of avoiding unnecessary calls to the logging objects when there are no handlers.  It may not seem like much, but it makes a big difference in overhead:
```python
>>> bacpypes.core._debug
1
```
Now the functions and methods in the module will generate debug output.  Unlike more generic formatters, the logging formatter in the debugging module knows how to "explode" the contents of objects:
```python
>>> bacpypes.core.run_once()
DEBUG:bacpypes.core.run_once:run_once
DEBUG:bacpypes.core.run_once:    - task, delta: None, None
```
Sending the debugging content someplace else is creating one of the standard handlers and passing that rather than letting ConsoleLogHandler use its own stream handler:
```python
handler = logging.handlers.RotatingFileHandler(
    "snork.txt", maxBytes=1024 * 1024, backupCount=10,
    )
snork = ConsoleLogHandler("bacpypes.core", handler=handler)
```
And you can pass the same handler to multiple ConsoleLogHandlers for debugging multiple modules, or create more than one of these to split things into multiple files.  You can also pass debugging options in as environment variables:
```bash
# debugging file can rotate, set the file size large to keep
# it from rotating a lot
export BACPYPES_MAXBYTES=10485760

# add the modules or classes that need debugging and redirect
# the output to the file
export BACPYPES_DEBUG=" \
    tests.test_service.helpers.ApplicationNetwork:$bugfile \
    tests.test_service.helpers.SnifferStateMachine:$bugfile \
    tests.state_machine:$bugfile \
    "
```

ChristianTremblay, 2020-09-15:
No activity for a long time. If help still required, please reopen

closed 2020-09-15 by ChristianTremblay