JoelBender/bacpypes #272: Exceptions generated by IOCB callbacks are silently ignored

zoopp, 2019-05-17:
I am using IOCBs for asynchronous request result handling and noticed that if the callback that gets called raises and exception then that exception is silently ignored (just noticed that further IOCBs aren't handled anymore by bacpypes as well). This gave me the impression that everything worked fine whereas in reality there was an issue which I was not aware of. Bellow is a sample application that can be used to reproduce the issue:

```python
import threading
import time

import bacpypes.core

from bacpypes.apdu import ReadPropertyRequest
from bacpypes.app import BIPSimpleApplication
from bacpypes.local.device import LocalDeviceObject
from bacpypes.pdu import Address
from bacpypes.iocb import IOCB

class App(BIPSimpleApplication):
    def __init__(self, local_address: Address):
        self._device_object = LocalDeviceObject(objectIdentifier=0xDEAD,
                                                vendorIdentifier=0xBEEF)
        BIPSimpleApplication.__init__(self, self._device_object, local_address)

    def repro_issue(self):
        request = ReadPropertyRequest(
            destination=Address('127.0.0.1'),
            objectIdentifier=0xDEAD,
            propertyIdentifier='objectName')
        iocb = IOCB(request)
        iocb.add_callback(self.cb_result)
        self.request_io(iocb)

    def cb_result(self, iocb: IOCB):
        print('Before raise')
        raise ValueError('What happened with this exception?')

app = App(Address('0.0.0.0'))
thread = threading.Thread(target=bacpypes.core.run,
                          daemon=True)

thread.start()
while True:
    time.sleep(1)
    print('Calling repro_issue')
    app.repro_issue()
```

I would propose to make the exception reach the user/developer so that they become immediately aware that there's an issue with their code.

JoelBender, 2019-05-21:
This is actually worse than what you've described because it can be replicated without anything other than the core:
```python
from bacpypes.debugging import bacpypes_debugging, ModuleLogger
from bacpypes.consolelogging import ArgumentParser

from bacpypes.core import run, deferred, stop

# some debugging
_debug = 0
_log = ModuleLogger(globals())

@bacpypes_debugging
def deathly_hallows():
    if _debug:
        deathly_hallows._debug("deathly_hallows")
    raise RuntimeError("cloak of invisibility")

# parse the command line arguments
args = ArgumentParser(description=__doc__).parse_args()

if _debug:
    _log.debug("initialization")
if _debug:
    _log.debug("    - args: %r", args)

deferred(deathly_hallows)
deferred(stop)

_log.debug("running")

run()

_log.debug("fini")
```
Notice that the function should be called and then `stop()`, but because of the exception it never gets that far.  If you run this with debugging turned on like `--debug bacpypes.core` then you can see that the exception is caught by the `except Exception as err:` statement in the `run()` function.

This was introduced while the library was being developed and there were socket and encoding/decoding errors that there was no way to predict and I couldn't seem to test for them all.  So this was a "stop gap" that hasn't gone away and has tripped up more than one BACpypes user.

I'm open to changing this to remove the `if _debug:` part for the next release to be more like:
```python
        except Exception as err:
            run._exception("an error has occurred: %s", err)
```
which, with enough advanced warning, shouldn't trip up existing code and help expose more problems that can generate better solutions.

ChristianTremblay, 2020-09-15:
Merged.... we'll close

closed 2020-09-15 by ChristianTremblay