JoelBender/bacpypes #274: Invalid syntax in BACread.py

Saravan95, 2019-05-27:
File "BACread.py", line 42
    def do_read(self, '192.168.0.3:47808/24' 852091 85):
                                           ^
SyntaxError: invalid syntax

Need HELP
ThankYou in Advance

JoelBender, 2019-05-28:
When you clone the repository there is a file in the root directory called [BACpypes~.ini](https://raw.githubusercontent.com/JoelBender/bacpypes/master/BACpypes~.ini), make a copy of the file and edit it to your local device settings.  The `objectName` is the name of your device, the `address` is the IPv4 address in CIDR notation, and the `objectIdentifier` is the instance number of your device object, also known as the "device instance number".  The `foreignBBMD` and `foreignTTL` are used for foreign device registration, which you probably won't need right away.

JoelBender, 2019-05-28:
By default the applications look in the current directory for the INI file, but most of them also take an `--ini` option which can point to the file, and in some cases it's easier to set an environment variable, so `BACPYPES_INI` can reference it.

closed 2020-09-15 by ChristianTremblay