JoelBender/bacpypes #275: BBMD configuration issue?

sirmuzz, 2019-05-28:
I am trying to test doing foreign device discovery using bacpypes BIPSForiegnApplicaiton.  I have a device that is on another network (166.55.99.150) this is a BASRouter.  The BASRouter has an MS/TP device it at 50:1 instance 1, 50:2 instance 2 and 50:3 instance 3.  When i specify the bbmd address as "166.55.99.150" and ttl 300 in the BIPSForigenApplication, i cannot get any response from the device.  Locally using the BIPSSimple i can see local bacnet devices.  I can using backstack or Yabe to see the remote devices without any issue.  Am i missing something simple?  I am able read and write properties to these devices, but cannot get the bbmd (whois) for discovery functionality to work.    I am fairly new to bacpypes and bacnet and appreciate your assistance.

JoelBender, 2019-05-29:
The router is also acting as the BBMD?  The first thing to check is that you are getting a registration acknowledgement, otherwise the BACpypes application drop attempts to send broadcast messages like Who-Is and Who-Is-Router-To-Network.  To debug this layer, run the application again and turn on debugging by adding `--debug bacpypes.bvllservice.BIPForeign` to the command line.  The debugging messages you see will mostly be _BACnet Virtual Link Layer_ PDUs and you can line up the pieces with _Annex J.5_, like `RegisterForeignDevice` and `Result`, hopefully with a result code zero.

sirmuzz, 2019-05-29:
Joel,
     Thank you for the information.  Yes the BasRouter is acting as a BBMD.  After further investigation this morning it appears my registration request was not completing before doing the who_is.   As i suspected your implementation is solid, i just had a brain fart on the registration process and waiting for it to complete.  Thank you for your time and help, and of course this great library.

tnesztler, 2019-06-11:
We are also trying to connect to ARCnet devices behind a BACnet IP router (ACL LGR250) w/o success (network 2473 and address 3 for example).
We are stuck using v0.16.7 at the moment (due to using it as part of the Volttron software package) however.
I have set the foreignBBDM IP as the router IP in the ini file and tried with your debugging settings. We get a `BIPForeign:confirmation` with a `bvlciResultCode = 48`. I couldn't find out what it means.
We've also modified a ReadPropertyForeign sample to work on v0.16.7. When we try to do a `read_prop 2473:3 device 2470013 objectName`, we get `noResponse`.
Do you know what could be the issue, if a BBMD configuration issue at all?

JoelBender, 2019-06-12:
That non-zero result code is a **Register-Foreign-Device NACK**, See _Clause J.2.1.1_.  Since the registration request failed, BACpypes will not send any broadcast messages, and because the stack doesn't know the router to network 2473, it can't broadcast a Who-Is-Rotuer-To-Network message to find it.

You may also be running into old code that simply broadcast the ReadPropertyRequest (this is _wrong_ by the way) expecting to clean up the routing tables in response.

You will have to figure out why your foreign device registration request is failing, but in the mean time you can update the routing table inside the `NetworkServiceAccessPoint` so it knows the address of the router.  I suspect that you want to do something like this...
```python
    x.nsap.add_router_references(x.nsap.adapters[0], Address("1.2.3.4"), [2473])
```
...but it's been a while!

tnesztler, 2019-06-12:
I was able to read data off of the ARCnet devices thanks to your help @JoelBender ! I was using v0.16.7 in this case.

For those who might come to this thread in the future:
I've used the `ReadProperty.py` sample and run `rtn 1.2.3.4 2473` (that's the `add_router_references` command) + `read 2473:3 device 2470013 objectList` for example.

I will still have to implement that into the Volttron software but that's the least of my problems.

closed 2020-09-15 by ChristianTremblay