JoelBender/bacpypes #277: How to flush BIPSimpleApplication.request_io queue?

orihime-kitajima, 2019-06-13:
I'm developping building simulator controller with BACnet / bacpypes.
I want to flush BIPSimpleApplication.request_io packet queue, before reset BACnet server.

How to flush queue which is used in BIPSimpleApplication.request_io ?

JoelBender, 2019-06-13:
The `ApplicationIOController` has a `SieveQueue` for each destination address.  These are instances of an `IOQController` which has an `abort()` method that pops each IOCB out of its queue, changes the state to aborted, and triggers it.

Try adding this function to `ApplicationIOController` which will implement the `abort()` method, you've probably already noticed that the more generic `IOController` it inherits from doesn't have an implementation:
```python
    def abort(self, err):
        if _debug: ApplicationIOController._debug("abort %r", err)

        # aborting some requests may trigger other requests
        while self.queue_by_address:
            local_copy = self.queue_by_address
            self.queue_by_address = {}
            if _debug: ApplicationIOController._debug("    - aborting: %r", local_copy)

            for queue in local_copy.values():
                queue.abort(err)
```
There should probably be some kind of guard for infinite loops.  Let me know if this works, or if it doesn't obviously, and I'll add it.

orihime-kitajima, 2019-06-14:
Thank you very much!

Your implementation helped solve my problem, and succeeded in resetting the packet queue.

closed 2019-06-14 by orihime-kitajima