JoelBender/bacpypes #278: Help Removing Write Callback

tbwebb2, 2019-06-17:
Thanks to everyone who has helped contribute to this awesome project! 

I am using Bacpypes to manipulate 800+ VAV setpoints on a large university campus for an energy research project. 

At this point I would like to speed up the Write process by removing the callback. Basically, I don't need Bacpypes to wait for the "Ack" packet to come back from the field device. Any help on achieving this would be greatly appreciate (I'm a Python beginner). Below is a relevant snippet of the code I'm using:

def writeToBACnetPoint(ipAddress, objectType, objectInstance, objectProperty, valueToWrite):
    # Write value to a BACnet point
    global globalValueToWrite
    globalValueToWrite = valueToWrite
    if float(valueToWrite) > 80 or float(valueToWrite) < 60:
        valueToWrite = 70
    print("Writing Value ", valueToWrite, "to ", ipAddress, " ", objectType, ":", objectInstance, " ", objectProperty)
    
    built_list = [(ipAddress, objectType + ":" + objectInstance, objectProperty)]
    bacpypesWriteToPoint(built_list)

def bacpypesWriteToPoint(bacpypesVAVPointList):
    global this_application
    global this_device
   
    # parse the command line arguments
    args = ConfigArgumentParser(description=__doc__).parse_args()

    this_application = WritePointsListApplication(bacpypesVAVPointList, this_device, args.ini.address)

    # fire off a request when the core has a chance
    deferred(this_application.next_request)
    
    enable_sleeping()

    run()

    this_application.close_socket()

class WritePointsListApplication(BIPSimpleApplication):
    def __init__(self, point_list, *args):
        BIPSimpleApplication.__init__(self, *args)

        # turn the point list into a queue
        self.point_queue = deque(point_list)

        # make a list of the response values
        self.response_values = []

    def next_request(self):
        global globalValueToWrite

        # check to see if we're done
        if not self.point_queue:
            if _debug: WritePointsListApplication._debug("    - done")
            stop()
            return

        # get the next request
        addr, obj_id, prop_id = self.point_queue.popleft()
        obj_id = ObjectIdentifier(obj_id).value
                
        value = str(globalValueToWrite)
        
        # get the datatype
        datatype = get_datatype(obj_id[0], prop_id)
        
        # change atomic values into something encodeable, null is a special case
        if (value == 'null'):
            value = Null()
            #print("Value Null")
        elif issubclass(datatype, AnyAtomic):
            dtype, dvalue = value.split(':', 1)
            #print("    - dtype, dvalue: %r, %r", dtype, dvalue)

            datatype = {
                'b': Boolean,
                'u': lambda x: Unsigned(int(x)),
                'i': lambda x: Integer(int(x)),
                'r': lambda x: Real(float(x)),
                'd': lambda x: Double(float(x)),
                'o': OctetString,
                'c': CharacterString,
                'bs': BitString,
                'date': Date,
                'time': Time,
                'id': ObjectIdentifier,
                }[dtype]
            #print("    - datatype: %r", datatype)

            value = datatype(dvalue)
            #print("    - value: %r", value)

        elif issubclass(datatype, Atomic):
            #print("Is sub class atomic")
            if datatype is Integer:
                value = int(value)
            elif datatype is Real:
                value = float(value)
            elif datatype is Unsigned:
                value = int(value)
            value = datatype(value)
        elif issubclass(datatype, Array) and (indx is not None):
            if indx == 0:
                value = Integer(value)
            elif issubclass(datatype.subtype, Atomic):
                value = datatype.subtype(value)
            elif not isinstance(value, datatype.subtype):
                raise TypeError("invalid result datatype, expecting %s" % (datatype.subtype.__name__,))
        elif not isinstance(value, datatype):
            raise TypeError("invalid result datatype, expecting %s" % (datatype.__name__,))
             
        # build a request
        request = WritePropertyRequest(
            objectIdentifier=obj_id,
            propertyIdentifier=prop_id,
            )

        request.pduDestination = Address(addr)
      
       # save the value
        request.propertyValue = Any()
        try:
            request.propertyValue.cast_in(value)
        except Exception as error:
            print("WriteProperty cast error: %r", error)

        # Make an IOCB
        iocb = IOCB(request)

        # set a callback for the response
        iocb.add_callback(self.complete_request)

        #give it to the application
        this_application.request_io(iocb)

    def complete_request(self, iocb):
        # fire off another request
        deferred(self.next_request)

tbwebb2, 2019-06-18:
I've realized that instead of removing the callback altogether, it may be easier to just make a very short timeout period. I'm struggling with how to do this, any help would be greatly appreciated.

JoelBender, 2019-06-19:
There are a few things you can do.  The `DeviceObject` has a property called `apduTimeout` which defaults to 3000ms.  If you're campus network is like mine it's relatively flat topology with just a few hops between your workstation/server and the field equipment, but the round trip time between requests and responses will vary depending on the load on the field bus (the MS/TP or ARCNET stuff downstream of a B/IP router) and the horsepower of the routers.

First, run your application and Wireshark filtering on BACnet traffic and save the pcap file, then clone the `ReadPropertySummaryFilter.py` application from the [bacpypes-pcap](https://github.com/JoelBender/bacpypes-pcap) project and update it for `WritePropertyRequest` content and see what kinds of timing you get.  Make sure you send the application to me to put in the project!

Second, adjust the `apduTimeout` so the application doesn't have to wait a long time for retry attempts, rinse and repeat.

Third, make a list of all of the destination addresses you are going to be writing to and sort them by network number and device address.  Make a queue of the requests for each network, then start a "thread" for each queue processing all of the requests in the related queue one at a time.  The effect is to multiplex all the requests that you can without overloading a device (in case there is more than one request for a device) and a network (for slow networks).  While there may be more than one outstanding request for a B/IP router if it has multiple downstream networks, routers don't keep application layer request state.

If there are just a few networks that are consistently slow or require retries, it is worth running some analysis tools on the field bus to see if you have too many VAV boxes on the same network, or there are termination problems, etc.  I'm not a big fan of ARCNET-to-ARCNET routers, but quite often they get added during the "value engineering" phase of a project.

tbwebb2, 2019-06-24:
Thanks for the response Joel. I ended up solving this using the iocb.set_timeout method. This way, I can reduce the timeout for writes, while allowing the read functions in the program to still have an appropriate timeout to receive the packet. 

It will take me some time to wrap my head around the threading you are talking about. I will have to get there eventually, as I have a vision of expanding this program to include fault detection & diagnostics, which will require many more Reads across the network than I am currently doing. Are you aware of anyone using Bacpypes for a custom FDD application?

ChristianTremblay, 2020-09-15:
> Are you aware of anyone using Bacpypes for a custom FDD application?

Yes.

closed 2020-09-15 by ChristianTremblay