JoelBender/bacpypes #279: ModuleNotFoundError using editable installation

fgervais, 2019-06-18:
I experienced this bug with bacpypes https://github.com/pypa/setuptools/issues/230#issuecomment-202914121

Basically when doing an editable installation it gives me a ModuleNotFoundError when trying to use bacpypes:

```
pip3 install -e ../bacpypes/
Obtaining file:///home/fgervais/4tb/bacpypes
Installing collected packages: bacpypes
  Found existing installation: bacpypes 0.17.6
    Uninstalling bacpypes-0.17.6:
      Successfully uninstalled bacpypes-0.17.6
  Running setup.py develop for bacpypes
Successfully installed bacpypes

python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import bacpypes
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'bacpypes'
>>> quit()
```

Using the fix proposed here https://github.com/pypa/setuptools/issues/230#issuecomment-202914121 works:

```
diff --git a/setup.py b/setup.py
index b2d4f99..6f3d106 100644
--- a/setup.py
+++ b/setup.py
@@ -60,7 +60,7 @@ setup(
         'bacpypes.service',
     ],
     package_dir={
-        'bacpypes': os.path.join(source_folder, 'bacpypes'),
+        '': os.path.join(source_folder, ''),
     },
     include_package_data=True,
     install_requires=requirements,
```

Would that be an acceptable fix? If so I can provide a PR for it.

JoelBender, 2019-06-19:
Is this an acceptable fix for many/all/some versions of `setuptools`?  I seem to have 38.4.0 for Python2.7, and Python3.5, 36.0.1 for Python3.6, 40.6.2 for Python3.7, and after the mess that was the transition from `distutils.core` I'm gun shy about this change.  I made a branch with this and it passes all the build/install tests I have, can you check it with different platforms?  This isn't going to change anything with respect to the `release_to_pypi.sh` script, correct?

fgervais, 2019-06-19:
Yes that is legitimate concerns which I cannot answer. I'll see if I can test on different platforms.

This is not a major problem as most people won't do this kind on install unless they are debugging something. Feel free to close this and at least there will be a trace here in case somebody else encounter the same problem. If it happens to more people in the future we could always reopen and think back about merging that in.

JoelBender, 2019-11-17:
This seems to work fine in **0.17.7** so we're going with it.  Thank you for your help and patch.

closed 2019-11-17 by JoelBender