JoelBender/bacpypes #280: WhoIsRouterToNetwork() Breaking in version 0.17.6 or Missing release notes??

bh4r4th, 2019-06-19:
Hi Joel,
I have tried to upgrade bacpypes version from 0.17.4 to 0.17.6 in my application which does a `WhoIsRouterToNetwork` discoveries. The application is failing to collect the results and couldn't trace the error. 
Working fine with 0.17.5. 
Can you add some release notes around the bug fixes applied in version 0.17.6, so that I will trace step by step to make sure my application supports this version. 
Thanks in Advance. 

Regards,
Bharath

JoelBender, 2019-06-19:
Hi Bharath,

I understand the release notes are missing and that's lame.  But in the mean time, I downloaded the tarballs of the two versions and unpacked them, then wrote this script to look for differences between the two versions showing the transition from 0.17.5 to 0.17.6:
```bash
#!/bin/bash

red=`tput setaf 1`
green=`tput setaf 2`
reset=`tput sgr0`
# echo "${red}red text ${green}green text${reset}"

files=`find bacpypes-0.17.6/py34/ | grep [.]py$`

for f in $files
do
    echo "${red}-----" $f "-----${reset}"
    diff ${f/0.17.6/0.17.5} $f
done
```
As a brief summary; `NetworkPortObject` was added, some segmentation window stuff in `appservice`, removed the default root log handler in `debugging.py`, added new classes `Unsigned8` and `Unsigned16`, added `SequenceOfAny`, and updated the foreign device code to be a little smarter.  There's nothing that should have changed the network discovery functionality.

JoelBender, 2019-06-30:
I've been hacking on #213 and this could solve the problems you are having, there are substantial changes to the `RouterInfoCache`.  Please test this branch and see if you have more success, otherwise we'll probably need to dig into Wireshark dumps.  The [BACpypes-pcap](https://github.com/JoelBender/bacpypes-pcap) project has some network filtering tools and it's pretty simple to add some additional ones for this problem.

bh4r4th, 2019-07-05:
Thanks @JoelBender. Awesome, Love your work and contribution. Will play with it next week or so. Currently I am on holiday. Definitely will give it a go after I am back from holidays.

bh4r4th, 2019-08-06:
I have used #213 as my bacpypes library and its now working as expected as per v0.17.4.
[image]

ChristianTremblay, 2019-08-12:
I’m using stage and WhoIsRouterToNetwork seems to work properly. I’m using a modified version of Network sevice élément that intercept responses. Is there any edge case or bug I could check for ?

JoelBender, 2019-11-17:
I'm closing this as resolved, please upgrade to 0.17.7 and check it again when you get a chance.  If this bug pops back up again please create another issue.

closed 2019-11-17 by JoelBender