JoelBender/bacpypes #281: Static Route

JoelBender, 2019-06-20:
There is an application that needs to direct traffic to a specific router to be forwarded on to a specific network.  This enhancement is to support "static routes" so the application can provide a remote broadcast or remote station address via a specific local station.  For example `2:5@1.2.3.4` will be sent as `2:5` is now via `1.2.3.4` as the router to network 2.  Similarly, `2:*@3.4.5.6:47809` will be a directed broadcast on network 2 via `3.4.5.6` port 47809.

sirmuzz, 2019-06-27:
Joel,  I use bacpypes to monitor some remote building devices and am constantly having to install/reinstall routes etc to specific networks, how is this enhancement going to help there?  I am very interested in this =)

JoelBender, 2019-06-27:
Checkout the `281-static-route` branch and install, then set `BACPYPES_ROUTEAWARE=True` in your environment.  Messages that you receive at the application layer that have gone through the network layer transformation (those that were routed to get to you) will have an addition `addrRoute` attribute that is an `Address` of the local station that delivered the message to you and will appear after the '@' in the string form of the address.  For example:

    DEBUG:__main__.WhoIsIAmApplication:indication <bacpypes.apdu.UnconfirmedCOVNotificationRequest(2) instance at 0x7f1ec3c60080>
        <bacpypes.apdu.UnconfirmedCOVNotificationRequest(2) instance at 0x7f1ec3c60080>
            pduSource = <RemoteStation 420:21@10.192.9.24>
            pduDestination = <GlobalBroadcast *:*>
            pduExpectingReply = False
            pduNetworkPriority = 0
            ...

This means that the message is a global broadcast sent by address 21 on network 420 and it arrived to you via a router at 10.192.9.24.

Messages that you send that are "route aware" will use the `addrRoute` as the local station address for other networks, bypassing the Who-Is-Router-To-Network hunting.  For example:

    $ BACPYPES_ROUTEAWARE=True python3.5 samples/ReadProperty.py
    > read 420:32@10.192.9.24 device:42032 objectName
    GatesHall.412.ZN141v
    > Exiting...

The request will be sent to 420:32 using 10.192.9.24 as the router.  This is specifically designed for situations where there could be another IPv4 device, maybe at 10.192.24.16, that is also presenting itself as a router to network 420, and therefore both IPv4 devices cannot be in the same BACnet intranet without all hell breaking loose.

You might also be interested in the [BACnet Masquerade Router](https://github.com/JoelBender/bmr) which should be able to mash two or more of these networks together with proxy devices on a virtual network.

_Note this branch is currently only in the py34 source tree._

JoelBender, 2019-11-17:
This code is now release in **0.17.7**, @sirmuzz if you have questions or problems please create another issue.

closed 2019-11-17 by JoelBender