JoelBender/bacpypes #282: WhoIs command only successful on a very specific subnet

pshank, 2019-06-21:
I should preface this with the fact that I don't necessarily think this is an issue, but something I am doing wrong/forgetting with the library. However, I have run into an interesting problem regarding the WhoIs command. 

Essentially, WhoIs is only working on a very specific subnet. I have a Gateway running the code, which is attached to a BACnet device via ethernet. This is a closed network, with a single cable running directly from the Gateway to the device. 

I have proven out that as long as my Gateway and the BACnet device are on the same subnet, the can communicate with each other (ping works, so does ReadPropertyMultiple). However, the ONLY way WhoIs is successful is if I have both devices set to 192.168.254.x. If I change any of the first 3 octets, the WhoIs command cannot find the device. As I mentioned, it has no problem with the ReadPropertyMultiple command, though. 

Is there some kind of... default setting and/or configuration that I may have done in the past but have since forgotten that may be effecting this? It is just very odd to me that it only likes that ONE subnet. 

Also to clarify, starting the WhoIs tool is not an issue (IE I am not getting any error messages when running 'python WhoIsIAm'.) It is just that sending the 'whois' command returns nothing.

JoelBender, 2019-06-22:
What you have forgotten is the concept of the _broadcast domain_, which is the set of all devices that are members of a domain that can receive a broadcast message.  For IPv4 messages, the broadcast domain is all of the devices the belong to the same subnet, which means they have the same value in the network portion of the address, which is specified by the subnet mask like **FF.FF.FF.00** (as hex octets) or **/24** in CIDR notation, the uppermost 24 bits on.

Who-Is broadcast messages are sent to the special address that has all one's in the host portion of the address of the sending device.  So if your source address is 192.168.0.10/24, the network is 192.168.0.X and the broadcast address is 192.168.0.255.  If your source address is 10.11.12.15/16, your network is 10.11.X.X and your broadcast address is 10.11.255.255. _You can have both IPv4 networks on the same Ethernet at the same time._

If you want a wider variety of network addresses, make your subnet bigger by making the network portion of the address smaller.  Setting both devices to the 192.168.0.0/16 range gives you 65534 addresses (zero is a very old broadcast address that isn't used any more) and 65535 is the broadcast address.  But even then, if you change one of the devices to 192.169.X.X/16 you'll no longer be in the same network.

pshank, 2019-06-24:
Thank you for the reply! While that makes sense, I must still be a little confused on the networking side of this.
For example, I would expect the following to work based on my understanding of what you said:

My source devices's IP address is 200.200.254.50, subnet set to 255.255.255.0
The BACpypes.ini file specifies the address/subnet as 200.200.254.50/24
Thus the network should be 200.200.254.X, with a Broadcast address of 200.200.254.255. (Running 'ifconfig' confirms this as the case, at least from the source device's perspective) So, I would expect that running "WhoIs" who broadcast the message to any device in the 200.200.254.X subnet.

The Device I am trying to discover has an IP of 200.200.254.101, subnet set to 255.255.255.0.
But this does not return anything when attempting to broadcast the WhoIs message.

ChristianTremblay, 2020-09-15:
Network issue. If help still required, please reopen

closed 2020-09-15 by ChristianTremblay