JoelBender/bacpypes #284: how to create virtual bacnet devices and implementing python

praveengowda1, 2019-07-30:
Hi @JoelBender  im very new to the bacnet protocol.I'm working in python , so i need help regarding python library called BACpypes. i already read BACpypes documentation on the internet and tried their examples but im not able to get the currect output. please help me with some examples like how to create virtual bacnet devices and how to perform read and write operation on which using python programing.also help me with some examples , thanks in advance,

JoelBender, 2019-07-31:
Based on what I know about the BACnet protocol tutorials available now on the Internet it's hard to gauge how much you know and what specifically you need help with.  Your understanding of how protocols are constructed, what the purpose of various layers are, what other protocols you have had experience with are all going to influence how you're going to get from where you are to where you want to be!

So begin with the end in mind... what specifically are you trying to do?

praveengowda1, 2019-07-31:
hi @JoelBender my task is to create one bacnet device virtually, and i need to read/write operation on that device using python library

JoelBender, 2019-08-01:
The **IP2VLANRouter.py** sample code is your answer.  When you run the application you are going to give it some parameters like this:
```bash
$ python IP2VLANRouter.py 192.168.0.12/24 10 20 --count 5
```
Assuming your local IP address is 192.168.0.12 and you're on a /24 network, this will turn your device into a router between network 10 and a virtual network 20, which will have 5 virtual devices on it with device instance numbers 2001 through 2005.  In BACpypes applications you'll see there addresses as `20:1`, other BACnet tools will use a different notation.  Reading and writing to those objects is shown in the **ReadWriteProperty.py** sample application.

damienpicard, 2019-10-10:
Dear Joel,

thank you for this useful post. Following your guidelines, I created the router:
`python samples\IP2VLANRouter.py 10.34.xxx.xxx:47809 10 20 --count 5`

I did not manage to do a device discovery without BBMD, but once I made one, I discovered them using DeviceDiscoveryForeign:
`python samples\BBMD.py 10.34.xxx.xxx:47811 10.34.xxx.xxx:47810 10.34.xxx.xxx:47809`
```
python samples\DeviceDiscoveryForeign.py --ini BACpypes.ini
> whois
> 2002 is at 20:2 named 'VLAN Node 2002'
2003 is at 20:3 named 'VLAN Node 2003'
2004 is at 20:4 named 'VLAN Node 2004'
2005 is at 20:5 named 'VLAN Node 2005'
2006 is at 20:6 named 'VLAN Node 2006'
```

However, I'm a bit loss with the addresses I get. How can I know use ReadProperty.py?
I have tried the following:
```
>python samples\ReadProperty.py --ini BACpypes.ini
> read 20:10.34.xxx.xxx device:2002 objectName
noResponse
> read 10.34.xxx.xxx/20:2 device:2002 objectName
noResponse
> read 20:2 device:2002 objectName
noResponse
```

Could you maybe help me out? I think that the problem is that I don't understand what the networks 10 and 20 are/means. I also have tried to get information about the router but I got an error message:
```
C:\work\python\bacpypes>python samples/WhoIsRouter.py --ini BACpypes.ini
> wirtn*
ERROR:bacpypes.consolecmd.ConsoleCmd:exception: AttributeError("'NoneType' object has no attribute 'process_npdu'",)
Traceback (most recent call last):
  File "C:\Python\lib\site-packages\bacpypes\consolecmd.py", line 95, in onecmd
    rslt = cmd.Cmd.onecmd(self, cmdString)
  File "C:\Python\lib\cmd.py", line 217, in onecmd
    return func(arg)
  File "samples/WhoIsRouter.py", line 105, in do_wirtn
    this_application.request(this_application.nsap.local_adapter, request)
  File "samples/WhoIsRouter.py", line 43, in request
    BIPNetworkApplication.request(self, adapter, npdu)
  File "C:\Python\lib\site-packages\bacpypes\comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "C:\Python\lib\site-packages\bacpypes\netservice.py", line 661, in sap_indication
    adapter.process_npdu(xpdu)
AttributeError: 'NoneType' object has no attribute 'process_npdu'
```

Thank you very much in advance

JoelBender, 2019-10-12:
Correct, you are missing some BACnet networking concepts, the introduction in _Clause 6
THE NETWORK LAYER_ in the standard is the definitive text.  BACnet intranets are not that much different than other kinds of intranets with the exception that each network can have a different address size.  For example, MS/TP and ARCNET use just a single octet (a.k.a. byte) while BACnet/IP and Ethernet use 6-octet addresses.

In BACpypes when the source and destination are on the same network the syntax is just the address like "12" or "192.168.2.3".  But when you (as a client) are sending a request to a device on a different network (as a server) you put the network number in front, like "20:12" or "20:192.168.2.3".

So in your case you have a client device on network 10, let's say 192.168.0.11, and the router between network 10 and 20 at 192.186.0.12, and a virtual device on the virtual network 20 at address 2.  You now have four interesting addresses in your BACnet network 10:192.168.0.11, 10:192.168.0.12, 20:1, and 20:2.  The 20:1 is the address of the router on the network.

Requests from you like `read 20:2 device:2002 objectName` will first go from you at 192.168.0.11 to the router at 192.168.0.12, then be repackaged by the router to say "this is from 10:192.168.0.11" and then sent to 2.  The device at 2 sends the response back to the router at 1 with a header that says "please forward this to 10:192.168.0.11".  The router repackages the message to say "this is from 20:2" and sends it to you.  So you see this response coming from the router at your IP network layer, but at the BACnet network layer you know it's from 20:2.

damienpicard, 2019-10-14:
Dear Joel,

thank you very much for these clarifications. So if I understand it correctly, following command
`python samples\IP2VLANRouter.py 10.34.xxx.xxx:47809 10 20 --count 5`
creates 5 devices on 20:10.34.xxx.xxx:47809 + a router between network 10 and 20.

Then, with following command:
`python samples\ReadProperty.py --ini BACpypes.ini`
with `address: 10:10.34.xxx.xxx:47809` in BACpypes.ini, I create a BACnet device on network 10.

But which command should I use to read?
```
> read 20:2 device:2002 objectName
noResponse
> rtn 10:10.34.xxx.xxx 20
> read 20:2 device:2002 objectName
noResponse
> read 20:10.34.xxx.xxx:47809 device:2002 objectName
noResponse
```

Thank you in advance!

damienpicard, 2019-10-15:
Hi Joel, 

could it be that not router was created?
```
C:\work\python\bacpypes>python samples/WhoIsRouter.py --ini BACpypes.ini
> wirtn*
ERROR:bacpypes.consolecmd.ConsoleCmd:exception: AttributeError("'NoneType' object has no attribute 'process_npdu'",)
Traceback (most recent call last):
  File "C:\Python\lib\site-packages\bacpypes\consolecmd.py", line 95, in onecmd
    rslt = cmd.Cmd.onecmd(self, cmdString)
  File "C:\Python\lib\cmd.py", line 217, in onecmd
    return func(arg)
  File "samples/WhoIsRouter.py", line 105, in do_wirtn
    this_application.request(this_application.nsap.local_adapter, request)
  File "samples/WhoIsRouter.py", line 43, in request
    BIPNetworkApplication.request(self, adapter, npdu)
  File "C:\Python\lib\site-packages\bacpypes\comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "C:\Python\lib\site-packages\bacpypes\netservice.py", line 661, in sap_indication
    adapter.process_npdu(xpdu)
AttributeError: 'NoneType' object has no attribute 'process_npdu'
>
```

damienpicard, 2019-10-17:
any hint? :)

JoelBender, 2019-10-17:
Ah!  This is a bug in the *master* branch but has been fixed in *stage* and other branches.  If you look at the end of this diff you can see the change that was made to the network service access point `bind()` function:
```
$ git diff master..stage -- py34/bacpypes/app.py
         # bind the NSAP to the stack, no network number
-        self.nsap.bind(self.bip)
+        self.nsap.bind(self.bip, address=self.localAddress)
```
So you can check out the stage branch, patch that line of code and re-install, or wait until the next release.

damienpicard, 2019-10-18:
Hi Joel,

I have replaced the whole bacpypes folder from `C:\Python\Lib\site-packages` to use the stage version.

I still get no response and now, it seems that I cannot use `rtn` anymore.
```
C:\work\python\bacpypes>python samples\ReadProperty.py --ini BACpypes.ini
> read 20:2 device:2002 objectName
noResponse
> rtn 10:172.16.xxx.xxx 20
exception: AttributeError("'NetworkServiceAccessPoint' object has no attribute 'add_router_references'",)
Traceback (most recent call last):
  File "C:\Python\lib\site-packages\bacpypes\consolecmd.py", line 95, in onecmd
    rslt = cmd.Cmd.onecmd(self, cmdString)
  File "C:\Python\lib\cmd.py", line 217, in onecmd
    return func(arg)
  File "samples\ReadProperty.py", line 127, in do_rtn
    this_application.nsap.add_router_references(None, router_address, network_list)
AttributeError: 'NetworkServiceAccessPoint' object has no attribute 'add_router_references'
```

However, the code `WhoIsRouter.py` works now (before it was not working, see higher):
```
C:\work\python\bacpypes>python samples\WhoIsRouter.py --ini BACpypes.ini
> wirtn *
> exit
```
So, no error message but also no router information...

Just to be sure, I also tried to use the previous bacpypes version and only adapting the lines that you indicated here above (@ 2 different places) of `app.py` but it did not work either.

So we are moving forward but I'm not yet there :)

Thank you

Kristoff6, 2019-10-18:
Hi Joel,

I'm following along. I did check out the stage branch, `self.nsap.bind(self.bip, address=self.localAddress)` is in py34/bacpypes/app.py, so installed that with `python setup.py install` on a clean virtual CentOS7 machine using Python 3.6.

Unfortunately I'm having similar issues as Damien, ie. I can find the virtual devices with a foreign device discovery as listed above. However I don't receive any response on `read 20:2 device:2002 objectName`. I don't receive any router info either.

I'm not sure how to get this working... Do you have an idea what's going on?

Thanks in advance!

JoelBender, 2019-10-24:
@damienpicard The `add_router_references()` function had been renamed to `update_router_references()` because the function could update existing references as well as add new ones and the sample code wasn't updated to reflect the change. The commit `ed68c57` fixes that in the `stage` branch, please pull and re-install.

JoelBender, 2019-10-24:
@Kristoff6 Please sync up with the stage branch, if it's still not working please start a new issue.

Kristoff6, 2019-11-05:
Ok, its working now. Thank you!

closed 2019-11-06 by JoelBender

chaoukic, 2020-01-03:
Hello @JoelBender , I have a question.  So I was introduced to bacpypes by a friend of mine and I wanted to test it out. 
I tried the following command you gave 
`python IP2VLANRouter.py myIp/24 10 20 --count 5` 

I then I wanted to check all the virtual devices that were created, so I ran the following command
(I changed the BACpypes.ini to have the same address as the entered in the first command, myIp/24)
`python samples\DeviceDiscoveryForeign.py --ini BACpypes.ini`
`> whois`
So I wanted to know why am I unable to see the virtual devices I created. 
Thanks in advance :)

JoelBender, 2020-01-03:
See gitter...

chaoukic, 2020-01-03:
Ye sorry ;P