JoelBender/bacpypes #286: Missing option to disable COV for some variables

fromeijn, 2019-07-31:
Hey Joel and other Contributors,

I'm working on an application where i need to mix objects with and without cov functionality. I can't find this option anywhere, am I missing something or is this a missing feature?

Keep up the good work!

JoelBender, 2019-07-31:
It's a missing feature, how do you envision this being supported?  There needs to be some code in `do_SubscribeCOVRequest` that makes the distinction between objects that support COV and those that don't.

How about a `_object_supports_cov` class attribute set to `True` for standard objects that COV is expected to be available if the device supports COV services (like AnalogValueObject) but `False` for the others (like DeviceObject).  You would be able to turn it on for your objects at the class level for your derived classes `RandomAnalogValueObject._object_supports_cov = True` or at the instance level `ravo1._object_supports_cov = False`.

fromeijn, 2019-07-31:
Thanks for your quick reply. Yea I think it would be a good idea to enable by default as you described. This way it is backwards compatible and add this option as well. Would you like to write that code or shall I have a go at it?

JoelBender, 2019-07-31:
Go for it, and extra credit for tests!  Please fork from the `stage` branch so the pull request is easier to squash-merge.

fromeijn, 2019-08-01:
Did some work, want to know what you think of it. Also tried to create a test for it, but didn't quite understand it (yet). Also need to make it available for py25 and py27, although it is probably copy paste. [https://github.com/fromeijn/bacpypes/tree/stage](url)

JoelBender, 2019-08-03:
I understand what you did, I created a slightly different version, see the `286-disable-cov` branch [1].  I turned the flag off by default, then turned in on for all of the classes that have a criteria defined in the `criteria_type_map`.  Derived classes (like the `RandomAnalogValueObject`) can turn it back off by providing it at the class level, or turn it on and off for specific instances (so `ravo1` can have it on but `ravo2` can have it off).

In the standard everything that is not in _Table 13-1_ is covered in _Table 13-1a_ or _Table 13-1a-2_ so you could ask for COV notifications for "weird" things like the description of the device object.  BACpypes doesn't support that now on the server side, so this isn't breaking any backward compatibility, but to be compliant this will have to change.

[1] this is also just in the py34 source tree

fromeijn, 2019-08-06:
Looking good, this is indeed even better! are you planning to merge this later on?