JoelBender/bacpypes #290: No module named 'bacpypes'

shurcer, 2019-10-07:
hi.
using Python 3.6.8 i get the following error:

Traceback (most recent call last):
  File "samples/Tutorial/WhoIsIAm.py", line 11, in <module>
    from bacpypes.debugging import bacpypes_debugging, ModuleLogger
ModuleNotFoundError: No module named 'bacpypes'

Where am I wrong?
Thanks

JoelBender, 2019-10-07:
This commonly occurs when installation process put the library in a different directory along the path than what you expected.  It looks like you are running the application from the GitHub directory you cloned, so you are running the application like this:
```
$ python3.6 samples/Tutorial/WhoIsIAm.py
```
You should install the library from PyPI like this:
```
$ python3.6 -m pip install bacpypes
```
Or like this:
```
$ python3.6 -m pip install --user bacpypes
```
Or to install it from the directory you cloned, run the "usual" build and install process:
```
$ python3.6 setup.py build install
```
And don't forget to make a copy of the `BACpypes~.ini` and adjust it to your local network setttings:
```
$ cp BACpypes~.ini BACpypes.ini
$ vi BACpypes.ini
```

ChristianTremblay, 2020-09-15:
Local Python install issue. No activity for a long time. 
If help still required, please reopen

closed 2020-09-15 by ChristianTremblay