JoelBender/bacpypes #292: Seemingly no router added with IP2VLANRouter.py

damienpicard, 2019-10-29:
I'm using following sample code to generate a test set-up:
```python samples\IP2VLANRouter.py 10.34.xxx.xxx:47809 10 20 --count 5```.

While 5 objects are created (I can see them using `DeviceDiscoveryForeign`, I do not manage to call them using `ReadProperty.py`:
```
> read 20:2 device:2002 objectName
noResponse
> rtn 10:10.34.xxx.xxx 20
> read 20:2 device:2002 objectName
noResponse
```

Could it be that no router have been added?
```
C:\work\python\bacpypes>python samples\WhoIsRouter.py --ini BACpypes.ini
> wirtn *
> exit
```

I have used `ed68c57` from the stage branch.

JoelBender, 2019-10-30:
You are running the `IP2VLANRouter.py` application without specifying a subnet size so the application has no idea what it should listen to for broadcast messages.  Try running it with `10.34.xxx.xxx/16:47809` and if you run `netstat --udp -a -n` and you'll see two sockets open, one is `10.34.xxx.xxx:47809` for inbound and outbound unicast traffic and another is `10.34.255.255:47809` listening for broadcast traffic.

When you are running the `WhoIsRouter.py` application you are probably using the "normal" BACnet port 47808, so you two have two more sockets open, `10.34.xxx.xxx:47808` and `10.34.255.255:47808`.  Your local broadcast request for Who-Is-Router-To-Network is being sent from `10.34.xxx.xxx:47808` to `10.34.255.255:47808`, so it won't be received by the router because the port is different.

Try `wirtn 10.34.xxx.xxx:47809`, sending a unicast message directly to the router and you should get a response.

Next, with the `ReadProperty.py` application, tell the network layer the address of the router to network 20 with the command `rtn 10.34.xxx.xxx:47809 20`.  Then the application won't try to hunt for it, and the read requests will be successful.

damienpicard, 2019-10-31:
Wonderful, it works. Thank you very much!

The only pity is that the device does not allow me to write a value on `analogValue:1` and that the device does not have a priorityArray. I can thus not check my whole code.

JoelBender, 2019-10-31:
I can clone that application to make the point writable and/or commandable, support COV notifications, etc., let me know.  The point of this particular sample is to present a little bit of how a gateway application from BACnet/IP to some other protocol that supports multiple devices (maybe a master/slave protocol like MODBUS) can present each of those as virtual BACnet devices.

damienpicard, 2019-10-31:
Hi Joel,

if you could add the following items such that I can automate the testing of my code it would be very helpful. I tried to do it myself but so far without success.

- Different types of devices: binaryOutput, binaryInput, analogOutput, analogInput, analogValue, 
- write and read access with priorityArrays
- (optional as I don't use that yet) COV

Thank you very much in advance!