JoelBender/bacpypes #294: How to read properties from BACpypes to YABE as a simulated BACnet device?

julia-tech, 2019-11-06:
Hi,

I would like to use YABE to read properties from BACpypes. So is there any simulated  BACnet device  in BACpypes could be read in the YABE? 

For example, if the BACnet device in BACpypes is a temperature simulator. Then I can read the properties such as set point  and temperature to YABE.

Thanks,
Julia

JoelBender, 2019-11-06:
[This sample](https://github.com/JoelBender/bacpypes/blob/master/samples/HandsOnLab/Sample4_RandomAnalogValueObject.py) is a BACnet device that has a collection of random analog value objects, every time you read the `presentValue` property it returns a new number in the range 0..100.

julia-tech, 2019-11-26:
I use this sample, but can I read the property one by one in different conditions instead of change the 'present value' together?

JoelBender, 2019-11-28:
What kinds of conditions would you like to have?  I guessing that you would like the application to switch between multiple "states" given some event and the values of the objects staying constant until the application switches to a new state.