JoelBender/bacpypes #296: Access variable part of packet APDU

celineirvene, 2019-11-07:
I'm using the bacpypes framework to parse various packet details from BACnet MSTP PCAPs into pandas data frames. So far I've had much success extracting info like packet number, timestamp, APDU service, APDU type, etc. 

But, now I want to dig deeper and extract info from the variable portion of the APDU. For example, the vendor ID and service parameters from unconfirmedPrivateTransferRequests or the object identifiers and list of values from confirmedCOVNotifications. This has been a much more daunting task and in my troubleshooting it seems that those parts of the packet are not being decoded in apdu.py

Am I missing something key, how can I decode the 'variable' portions of packet APDUs? And if they are being decoded somewhere, how can I access this data? 

Thanks for you help.

JoelBender, 2019-11-08:
The `analysis` module is your friend.  You have already parsed the NPCI header and you know that you have an APDU (because the `npduNetMessage` is None), and you have decoded it as a generic APDU.  The next layer is a specific kind of APDU, like a `ConfirmedRequestSequence` or `ComplexAckSequence` that all inherit from `APCISequence`.  There is a `decode()` function in that class with this bit of code:
```
        # create a tag list and decode the rest of the data
        self._tag_list = TagList()
        self._tag_list.decode(apdu)
        if _debug: APCISequence._debug("    - tag list: %r", self._tag_list)
```
which turns the "variable encoding part" into a list of tags.  By turning on debugging on that class with a `--debug bacpypes.apdu.APCISequence` it will show you the list of tags just before they are "consumed" by the decoding of the sequence.  The mapping of the tags into Python primitive data (like an application encoded unsigned integer into a Python int) is handled by the `Sequence` class in the `constructeddata` module, which ends up being recursive thanks to sequences that contain other sequences, lists/arrays of stuff, etc.

If you have an MS/TP pcap file you wouldn't mind sharing I could add an `mstp` module with the appropriate link layer classes and update the `analysis` module to decode them.  Then all the applications in the [BACpypes-pcap](https://github.com/JoelBender/bacpypes-pcap) would work on these kinds of captures!

ChristianTremblay, 2019-11-08:
That would be nice... and I guess using pyserial for sniffing could work
and we would have something useful.

ChristianTremblay, 2020-09-15:
Added tag. Closing for now as not planned immediately.
Also, no activity for a long time.

closed 2020-09-15 by ChristianTremblay