JoelBender/bacpypes #298: Samples - BBMD/Whois usage

noenthu, 2019-11-15:
I am trying to setup a test environment using virtualbox vms with 2 subnets, BBMDs on each subnet and then a whoisIam application running on both subnets.

Subnet 1 - 10.1.1.1/24
WhoIsIam app1 - 10.1.1.2/24
BBMD sample app - 10.1.1.3/24 with bbmdt - 10.1.2.3/24

Subnet2 - 10.1.2.1/24
WhoIsIam app2 - 10.1.2.2/24
BBMD sample app - 10.1.2.3/24 with bbmdt - 10.1.1.3/24

Using the samples/BBMD.py I run the BBMD sample app on one vm in both subnets
Using another vm in each subnet I run the WhoIsIAM app and perform whois.
The two WhoIsIAM devices dont respond to each other.

If I do a direct whois from subnet 1 to subnet2 device it works

From 10.1.1.2/24 perform whois to 10.1.2.2 returns an IAM and vice versa but not with the BBMD sample.

Any input on how to setup a sample BBMD in each subnet with the remote BBMD added to its distribution table to allow cross subnet Whois to work?

JoelBender, 2019-11-16:
The BDT uses /24 entries which is "one hop" broadcasting which is typically blocked by routers.  Use /32 entries which will mean "two hop", the broadcast messages will be received by one, forwarded to the other, then re-broadcast.  Also make sure that the BBMD has itself as the first BDT entry and then it's peer:

    $ python samples/BBMD.py 10.1.1.2/24 10.1.1.2/32 10.1.2.2/32

The original design of BACnet/IP was that all BBMD's had identical BDT's because that fits the rest of the BACnet design, a broadcast on a network is delivered to every device on the network.  But of course that doesn't scale very well, so most folks use a "split horizon" like backbone networks for a building can see the central data center, and the data center can see all of the buildings, but the buildings can't see each other.

noenthu, 2019-11-18:
@JoelBender Thanks a lot. That did the trick. 
I was using 5 vms (2 on each subnet) and 1 vm with both interface/cards with 
net.ipv4.ip_forward=1 set on /etc/systctl.conf to allow network traffic through between the 2 interfaces.

closed 2019-11-18 by noenthu