JoelBender/bacpypes #299: Unhandled TypeError in case of segmented confirmed request message transmission

zoopp, 2019-11-19:
I came across an internal error that is raised in the following scenario:

* Issue a segmented confirmed request.
* Segment 0 is sent to the server and the segmented request timer is started.
* Timer expires before the server answers.

In this situation the ClientSSM will attempt to send the segments in the window again until the number of retries is exhausted BUT since there was no confirmation received, `self.actualWindowSize` has not been set and this leads to the TypeError observed in logs bellow.
```python
[2019-11-19 13:39:24,443] DEBUG:bacpypes.appservice.ClientSSM: process_task
[2019-11-19 13:39:24,444] DEBUG:bacpypes.appservice.ClientSSM: await_confirmation_timeout
[2019-11-19 13:39:24,444] DEBUG:bacpypes.appservice.ClientSSM:     - no response, try again (0 < 3)
[2019-11-19 13:39:24,446] DEBUG:bacpypes.appservice.ClientSSM: indication <bacpypes.apdu.ConfirmedRequestPDU(14,122) instance at 0x73defcd0>
[2019-11-19 13:39:24,447] DEBUG:bacpypes.appservice.SSM: set_segmentation_context <bacpypes.apdu.ConfirmedRequestPDU(14,122) instance at 0x73defcd0>
[2019-11-19 13:39:24,447] DEBUG:bacpypes.appservice.ClientSSM:     - segment size: 206
[2019-11-19 13:39:24,448] DEBUG:bacpypes.appservice.ClientSSM:     - invoke ID: 122
[2019-11-19 13:39:24,449] DEBUG:bacpypes.appservice.ClientSSM:     - segment count: 18
[2019-11-19 13:39:24,450] DEBUG:bacpypes.appservice.ClientSSM:     - server doesn't say maximum number of segments
[2019-11-19 13:39:24,451] DEBUG:bacpypes.appservice.ClientSSM: set_state 1 (SEGMENTED_REQUEST) timer=1500
[2019-11-19 13:39:24,452] DEBUG:bacpypes.appservice.SSM: set_state 1 (SEGMENTED_REQUEST) timer=1500
[2019-11-19 13:39:24,453] DEBUG:bacpypes.appservice.SSM: stop_timer
[2019-11-19 13:39:24,454] DEBUG:bacpypes.appservice.SSM: start_timer 1500
[2019-11-19 13:39:24,455] DEBUG:bacpypes.appservice.SSM: get_segment 0
[2019-11-19 13:39:24,456] DEBUG:bacpypes.appservice.SSM:     - confirmed request context
[2019-11-19 13:39:24,457] DEBUG:bacpypes.appservice.SSM:     - segmented response accepted: True
[2019-11-19 13:39:24,457] DEBUG:bacpypes.appservice.SSM:     - proposedWindowSize: 2
[2019-11-19 13:39:24,458] DEBUG:bacpypes.appservice.ClientSSM: request <bacpypes.apdu.ConfirmedRequestPDU(14,122) instance at 0x74f45df0>
[2019-11-19 13:39:25,956] DEBUG:bacpypes.appservice.ClientSSM: process_task
[2019-11-19 13:39:25,957] DEBUG:bacpypes.appservice.ClientSSM: segmented_request_timeout
[2019-11-19 13:39:25,958] DEBUG:bacpypes.appservice.ClientSSM:     - retry segmented request
[2019-11-19 13:39:25,958] DEBUG:bacpypes.appservice.SSM: start_timer 1500
[2019-11-19 13:39:25,959] DEBUG:bacpypes.appservice.SSM: fill_window 0
[2019-11-19 13:39:25,960] DEBUG:bacpypes.appservice.SSM:     - actualWindowSize: None
[2019-11-19 13:39:25,962] ERROR:bacpypes.core.run: an error has occurred: 'NoneType' object cannot be interpreted as an integer
Traceback (most recent call last):
  File "/home/mihaib/.local/share/virtualenvs/cbms-bacnetclient-tBbclg1q/lib/python3.7/site-packages/bacpypes/core.py", line 134, in run
    taskManager.process_task(task)
  File "/home/mihaib/.local/share/virtualenvs/cbms-bacnetclient-tBbclg1q/lib/python3.7/site-packages/bacpypes/task.py", line 376, in process_task
    task.process_task()
  File "/home/mihaib/.local/share/virtualenvs/cbms-bacnetclient-tBbclg1q/lib/python3.7/site-packages/bacpypes/appservice.py", line 421, in process_task
    self.segmented_request_timeout()
  File "/home/mihaib/.local/share/virtualenvs/cbms-bacnetclient-tBbclg1q/lib/python3.7/site-packages/bacpypes/appservice.py", line 533, in segmented_request_timeout
    self.fill_window(self.initialSequenceNumber)
  File "/home/mihaib/.local/share/virtualenvs/cbms-bacnetclient-tBbclg1q/lib/python3.7/site-packages/bacpypes/appservice.py", line 239, in fill_window
    for ix in range(self.actualWindowSize):
TypeError: 'NoneType' object cannot be interpreted as an integer
```
Quoting the BACnet standard bellow (emphasis mine):

> 5.3.2 Segmented Confirmed-Request Message Transmission
> 
> **Before sending the first segment of a segmented Confirmed-Request-PDU, a client device shall choose a Proposed Window Size to indicate the maximum number of message segments it is prepared to transmit before it must receive a SegmentACK.** The means of choosing the Proposed Window Size are a local matter, except that the value shall be in the range 1 to 127, inclusive. The Proposed Window Size shall be carried by the parameter of that name in each segment of the Confirmed-Request-PDU. The value of Proposed Window Size shall be the same in each segment of the Confirmed-Request-PDU.
> 
> Upon transmitting the first segment of a Confirmed-Request message, a client device shall start a timer that indicates the length of time this message segment has been outstanding. The timer shall be canceled upon the receipt of a Reject, Abort, or SegmentACK APDU for the outstanding Confirmed-Request message segment. If the timer exceeds the value of the APDU_Segment_Timeout property in the client's Device object, then the segment shall be retransmitted and the timer shall be reset to zero. **All retransmitted segments shall follow this same procedure until the message segment has been retransmitted the number of times indicated in the Number_Of_APDU_Retries property of the client's Device object**...

So, if I am interpreting this correctly, in this particular situation we should re-attempt to send the first segment with our proposed window instead.

zoopp, 2019-11-20:
I proposed a fix for this in https://github.com/JoelBender/bacpypes/pull/301.

JoelBender, 2019-11-20:
I merged this into [stage](https://github.com/JoelBender/bacpypes/tree/stage) (after my usual bit of wandering around git branches).  Thank you!

closed 2019-11-20 by JoelBender