JoelBender/bacpypes #304: AnalogValueObject - PropertyList Example / covIncrement

chrissli, 2019-11-22:
Hi Joel,

I'm currently trying to setup a device for BACnet using your library. Since I've haven't done anything to big with python I encountered some minor problems. I'm currently stuck with setting the "propertyList" on a AnalogValueObject.
I tried setting it with the property name and ID but still no effect:

```
temperature = AnalogValueObject (
	objectIdentifier = ('analogValue',1),
	objectName = moduleName,
	presentValue = value,
	statusFlags = [0, 0, 0, 0],
	covIncrement = 1.0,
	outOfService = False,
	eventState = 'normal',
	units = 'degreesCelsius',
	propertyList = ['units',117]
)
```

also in this style:

```
temperature = AnalogValueObject(
	objectIdentifier = ('analogValue',1),
	objectName = moduleName,
	presentValue = value,
	statusFlags = [0, 0, 0, 0],
	covIncrement = 1.0,
	outOfService = False,
	eventState = 'normal',
	units = 'degreesCelsius',
	propertyList = ['units',117]
)
```

Could you please provide an example how propertyList is setup?

Also, how does covIncrement affect the refresh duration? I tried setting it to 0.5 but the values still refresh only once per minute.
For testing I used "YABE - Yet Another Bacnet Explorer"

JoelBender, 2019-11-23:
The property list is an array of property identifiers, so you can build it like this:
```
from bacpypes.basetypes import PropertyIdentifier
from bacpypes.constructeddata import ArrayOf

# just to be friendly
ArrayOfPropertyIdentifier = ArrayOf(PropertyIdentifier)

property_list = ArrayOfPropertyIdentifier(['statusFlags', 'units'])

# property_list.debug_contents()
#     length = 2
#     [1] = 'statusFlags'
#     [2] = 'units'
```
And then pass it to the object:
```
from bacpypes.object import AnalogValueObject

temperature = AnalogValueObject(
	objectIdentifier = ('analogValue',1),
	objectName = 'some.object',
	presentValue = 75.3,
	statusFlags = [0, 0, 0, 0],
	covIncrement = 1.0,
	outOfService = False,
	eventState = 'normal',
	propertyList = property_list
)
```
You could also use a special mix-in class:
```
from bacpypes.object import register_object_type
from bacpypes.object import AnalogValueObject as _AnalogValueObject

from bacpypes.local.object import CurrentPropertyListMixIn

# registration is required
@register_object_type(vendor_id=999)
class AnalogValueObject(CurrentPropertyListMixIn, _AnalogValueObject):
    pass

temperature = AnalogValueObject(
	objectIdentifier = ('analogValue',1),
	objectName = 'some.object',
	presentValue = 75.3,
	statusFlags = [0, 0, 0, 0],
	covIncrement = 1.0,
	outOfService = False,
	eventState = 'normal',
)

# print out the array
# temperature.propertyList.debug_contents()

# now add some units
temperature.units = 'degreesCelsius'

# see it updated
temperature.propertyList.debug_contents()
```

JoelBender, 2019-11-23:
The COV subscription lifetime is used to determine how long to keep sending updates to the client, and the COV increment is used to determine if the value has changed enough to warrant sending another notification to the subscribed client(s).

chrissli, 2019-11-25:
Hi Joel,

thank you very much and sorry for the inconveniene.
I'm sure this will help me.
Thanks a lot for your hard work and efforts with bacpypes.

closed 2019-11-25 by chrissli