JoelBender/bacpypes #305: Pending confirmed request wrongfully marked as completed when issuing an unconfirmed request

zoopp, 2019-11-25:
I came across a situation in which pending active confirmed requests will be wrongfully marked as completed if an unconfirmed request is issued to the same device while waiting for a response. I isolated the issue in the following sample:

```python
import bacpypes.core

from bacpypes.apdu import ReadPropertyRequest
from bacpypes.app import BIPSimpleApplication
from bacpypes.iocb import IOCB
from bacpypes.local.device import LocalDeviceObject
from bacpypes.pdu import Address

device_object  = LocalDeviceObject(modelName='UT v0.1337',
                                   objectIdentifier=0xD00D,
                                   objectName='BACnet Device Object',
                                   systemStatus='operational',
                                   vendorIdentifier=0xB00F,
                                   vendorName='UTs INC',
                                   apduTimeout=5000,
                                   apduSegmentTimeout=5000,
                                   numberOfApduRetries=0)
application = BIPSimpleApplication(device_object, f'0.0.0.0:{0xBAC0}')

def do_request():
    def handle_result(result):
        print(f'Handling ReadPropertyRequest: {type(result)}, {vars(result)}')
        if all(x is None for x in [result.ioResponse, result.ioError]):
            raise ValueError

    print('Doing ReadPropertyRequest..')

    # Destination doesn't matter as long as we don't get a response before
    # doing the WhoIs request
    request = ReadPropertyRequest(
        destination=Address('123.123.123.123:12312'),
        objectIdentifier=('device', 123),
        propertyIdentifier='objectName'
    )
    iocb = IOCB(request)
    iocb.add_callback(handle_result)
    application.request_io(iocb)
    print('ReadPropertyRequest enqueued..')

def do_who_is():
    print('Doing targeted WhoIs..')
    application.who_is(address=Address('123.123.123.123:12312'))

bacpypes.core.deferred(do_request)
bacpypes.core.deferred(do_who_is)
bacpypes.core.run()
```

Which produces the following output:

```python
Doing ReadPropertyRequest..
ReadPropertyRequest enqueued..
Doing targeted WhoIs..
Handling ReadPropertyRequest: <class 'bacpypes.iocb.IOCB'>, {'ioID': 1, 'args': (<bacpypes.apdu.ReadPropertyRequest(12,1) instance at 0x7f2f365c10d0>,), 'kwargs': {}, 'ioState': 3, 'ioResponse': None, 'ioError': None, 'ioController': <bacpypes.iocb.SieveQueue object at 0x7f2f365c11c0>, 'ioComplete': <threading.Event object at 0x7f2f365c10a0>, 'ioCallback': [(<function do_request.<locals>.handle_result at 0x7f2f365b7700>, (), {})], 'ioQueue': None, 'ioPriority': 0, 'ioTimeout': None}
an error has occurred: 
Traceback (most recent call last):
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/core.py", line 165, in run
    fn(*args, **kwargs)
  File "sample.py", line 45, in do_who_is
    application.who_is(address=Address('123.123.123.123:12312'))
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/service/device.py", line 72, in who_is
    self.request(whoIs)
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/app.py", line 474, in request
    self._app_complete(apdu.pduDestination, None)
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/app.py", line 454, in _app_complete
    queue.complete_io(queue.active_iocb, apdu)
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/iocb.py", line 753, in complete_io
    IOController.complete_io(self, iocb, msg)
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/iocb.py", line 623, in complete_io
    iocb.trigger()
  File "/home/mihai/.virtualenvs/bacpypes-mihai/src/bacpypes/py34/bacpypes/iocb.py", line 172, in trigger
    fn(self, *args, **kwargs)
  File "sample.py", line 26, in handle_result
    raise ValueError
ValueError
```

JoelBender, 2019-11-26:
The `ApplicationIOController.request()` function is assuming that it is being called from the `SieveQueue.process_io()` instance specific to that destination address.  I renamed the `request()` function to `_app_request()` which fixes this problem.

I also put in an additional guard so that if `request()` is called with a confirmed service it raises an error to use an IOCB, otherwise when the ack/error/abort comes back it might not be for the active request because it wasn't queued properly. I had to adjust the tests (the `ApplicationStateMachine.send()` function) to match the new requirement.

Please test [this branch](https://github.com/JoelBender/bacpypes/tree/305-overlapping-requests).

zoopp, 2019-11-26:
I ran my UTs with this branch as well s testing some things with a real environment and things work okay. I'm unable to reproduce the issue with your fix.

zoopp, 2020-01-16:
I've been running my app with this fix for a while and haven't encountered any issues. I believe it should be safe to merge.

closed 2020-01-16 by JoelBender