JoelBender/bacpypes #306: How to read the properties one by one instead of changing all the

julia-tech, 2019-11-26:
Hi,

I use the code 

# make some random input objects

    for i in range(1, RANDOM_OBJECT_COUNT+1):
        ravo = RandomAnalogValueObject(
            objectIdentifier=('analogValue', i),
            objectName='Random-%d' % (i,),
            )
        _log.debug("    - ravo: %r", ravo)
        this_application.add_object(ravo)

but the present valye changed together for all the object, the value can be changed using the following code:

# return a random value
        value = random.random() * 100.0
        if _debug: RandomValueProperty._debug("    - value: %r", value)
        return value

can I return different value to different object each time? for example, for object1 , I can return 'present value = 21', and at the same time for object 2, I can return  'present value = 24' etc.

Thanks,
Julia

JoelBender, 2019-11-28:
I guess I don't understand, with this sample every time you read the present value of one of the analog value objects you always get a new value.  If you would like something that supports reading the values of two or more objects in one request, the [ReadPropertyMultipleServer.py](https://github.com/JoelBender/bacpypes/blob/master/samples/ReadPropertyMultipleServer.py) sample application can do that.

ChristianTremblay, 2020-09-15:
No activity for a long time. If help still needed, please reopen

closed 2020-09-15 by ChristianTremblay