JoelBender/bacpypes #307: NameError: name 'cov_detection' is not defined

hugovoerman, 2019-12-13:
Tool used: YABE
Version BACpypes: 0.17.7

Scenario:
View device object -> no problem
Subscribe to AnalogValue
View device object -> stackdump.

Analysis:

In service/cov.py (line 555) the call to
            cov_subscription = COVSubscription(
                recipient=recipient_process,
                monitoredPropertyReference=ObjectPropertyReference(
                    objectIdentifier=cov.obj_id,
                    propertyIdentifier=cov_detection.monitored_property_reference,
                    ),
                issueConfirmedNotifications=cov.confirmed,
                timeRemaining=time_remaining,
                )

uses cov_detection which is not initialized/present. 

Error / Stackdump:

> exception: NameError("name 'cov_detection' is not defined")
Traceback (most recent call last):
  File "C:\...\Python\Python37-32\lib\site-packages\bacpypes\app.py", line 382, in indication
    helperFn(apdu)
  File "C:\...\Python\Python37-32\lib\site-packages\bacpypes\service\object.py", line 309, in do_ReadPropertyMultipleRequest
    read_access_result_element = read_property_to_result_element(obj, propId, propertyArrayIndex)
  File "C:\...\Python\Python37-32\lib\site-packages\bacpypes\service\object.py", line 207, in read_property_to_result_element
    read_result.propertyValue = read_property_to_any(obj, propertyIdentifier, propertyArrayIndex)
  File "C:\...\Python\Python37-32\lib\site-packages\bacpypes\service\object.py", line 161, in read_property_to_any
    value = obj.ReadProperty(propertyIdentifier, propertyArrayIndex)
  File "C:\...\Python\Python37-32\lib\site-packages\bacpypes\object.py", line 598, in ReadProperty
    return prop.ReadProperty(self, arrayIndex)
  File "C:\...\Python\Python37-32\lib\site-packages\bacpypes\service\cov.py", line 559, in ReadProperty
    propertyIdentifier=cov_detection.monitored_property_reference,
NameError: name 'cov_detection' is not defined

Hopefully this can be solved.

JoelBender, 2019-12-14:
Humph.  The subscription object doesn't have a reference to the `COVDetection` instance, that has the reference to the monitored property, which has a value provided by one of the subclasses.  The question for me is; should the `Subscription` instance be given a reference to the monitored property when it's created (line 737) by being given the `cov_detection` instance, or just the `cov_detection.monitored_property_reference` value?  At some point it would be nice to support `SubscribeCOVPropertyRequest`.

Of course it can be solved, it's just code :-)

hugovoerman, 2019-12-17:
Hi Joel,

Just to check, do you expect an answer from me on your question: "The question for me is; should the Subscription instance be given a reference to the monitored property when it's created (line 737) by being given the cov_detection instance, or just the cov_detection.monitored_property_reference value?"

Please let me know if there's anything I can do.

JoelBender, 2019-12-18:
LOL - no, it was more thinking out loud.  There is a new branch for this fix, please test it and close the issue if it works and I'll roll it into stage.

hugovoerman, 2019-12-18:
Hi Joel,

I copied https://raw.githubusercontent.com/JoelBender/bacpypes/ad8449487ffbb16f5cde832280578f3c58537ee8/py34/bacpypes/service/cov.py into my install of 0.17.7.

Doing the same scenario now gives:

exception: TypeError('invalid result datatype, expecting ListOfCOVSubscription and got list')
Traceback (most recent call last):
  File "C:\Users\...\Python\Python37-32\lib\site-packages\bacpypes\app.py", line 382, in indication
    helperFn(apdu)
  File "C:\Users\...\Python\Python37-32\lib\site-packages\bacpypes\service\object.py", line 309, in do_ReadPropertyMultipleRequest
    read_access_result_element = read_property_to_result_element(obj, propId, propertyArrayIndex)
  File "C:\Users\...\Python\Python37-32\lib\site-packages\bacpypes\service\object.py", line 207, in read_property_to_result_element
    read_result.propertyValue = read_property_to_any(obj, propertyIdentifier, propertyArrayIndex)
  File "C:\Users\...\Python\Python37-32\lib\site-packages\bacpypes\service\object.py", line 179, in read_property_to_any
    % (datatype.__name__, type(value).__name__))
TypeError: invalid result datatype, expecting ListOfCOVSubscription and got list

JoelBender, 2019-12-19:
I went through a similar process; downloaded and unpacked bacpypes-0.17.7.tar.gz into a new directory, installed it into a new a virtual environment, and I could reproduce the `TypeError: invalid constructor datatype` error that you get when there are no active subscriptions, a related bug.

I copied the `cov.py` file from the `307-cov-detection-not-defined` branch into the `py34` source directory and ran `setup.py build install` and the code works fine, so something else is amiss.  One of the things that trips me up every once in a while is that the cached version doesn't always get updated the way I think it should during the build/install process, so I have a "purge" script that I run to make sure old stuff is gone.

I'm raising the "help wanted" flag on this one.

hugovoerman, 2019-12-19:
Hi Joel,

I will give it another try tomorrow and will let you know the result.

hugovoerman, 2020-01-27:
I found some time to test it again, but without success, unfortunately.

From a clean system I removed the globally installed bacpypes, got the .gz, unpacked it, put the new cov.py in its place and ran the setup. Actually just like you described above. On my side I still get the Exception. Exactly the same as I reported 18 Dec 2019.

Is there any progress on this meanwhile? Has someone taken the help wanted flag? Or will it 'hang' until you/I find time to debug it?

JoelBender, 2020-01-29:
Something ain't right.  I'm merging the branch and it will be in the next release, probably tomorrow.  You'll be able to **pip install --upgrade bacpypes** and it should work.

hugovoerman, 2020-02-01:
Hi Joel,

I'm very sorry that I must report that with 0.18.0 I get the following:

exception: TypeError('invalid result datatype, expecting ListOfCOVSubscription and got list')
Traceback (most recent call last):
  File "C:\Users\<user>\AppData\Local\Programs\Python\Python38\lib\site-packages\bacpypes\app.py", line 382, in indication
    helperFn(apdu)
  File "C:\Users\<user>\AppData\Local\Programs\Python\Python38\lib\site-packages\bacpypes\service\object.py", line 309, in do_ReadPropertyMultipleRequest
    read_access_result_element = read_property_to_result_element(obj, propId, propertyArrayIndex)
  File "C:\Users\<user>\AppData\Local\Programs\Python\Python38\lib\site-packages\bacpypes\service\object.py", line 207, in read_property_to_result_element
    read_result.propertyValue = read_property_to_any(obj, propertyIdentifier, propertyArrayIndex)
  File "C:\Users\<user>\AppData\Local\Programs\Python\Python38\lib\site-packages\bacpypes\service\object.py", line 178, in read_property_to_any
    raise TypeError("invalid result datatype, expecting %s and got %s" \
TypeError: invalid result datatype, expecting ListOfCOVSubscription and got list

Is there anything I can try to do to help us solve this issue?

This error occurs when I reqeust to read the device object by clicking on it in YABE. This device has active subscriptions (the bacnet server I made with bacpypes).

JoelBender, 2020-02-04:
Ok, back in my queue.  :stuck_out_tongue_closed_eyes: I'll let you know what I find.

hugovoerman, 2020-02-08:
Hi Joel,

I think I may have a lead for you. The change you made in cov.py were actually two changes. See: https://github.com/JoelBender/bacpypes/commit/ad8449487ffbb16f5cde832280578f3c58537ee8

I reverted the first change, so at line 527 of cov.py I changed
 cov_subscriptions = []
back to
cov_subscriptions = ListOf(COVSubscription)()

and now all problems are gone on my side!

The Bacnet Property Active Cov Subscriptions now shows a nice array all subscriptions.

I hope this helps you to get this out of the way.

Have a nice weekend,

Hugo

bGridSven, 2020-03-19:
I am experiencing the same issue. I'm using bacpypes 0.18.0

JoelBender, 2020-03-20:
I merged in the stage branch so it picked up the other changes that have been made, applied the patch that @hugovoerman made, and everything seems to be fine.  I want to add a test for this, probably in test_cov_bv.py and verify it before merging this back into stage and releasing it.

JoelBender, 2020-04-22:
This is finally solved.  The `ReadProperty` function wasn't accepting "Pythonic" lists as acceptable things to cast into the correct type.  For example, if the property was an `ArrayOf(Integer)` then the property value `[1, 2, 3]` should have been accepted and in some cases it was, but it got tripped up in other cases.  Commit 3960095ac191ef1bdcd906cbf953907c38480d35 finally fixed it in the correct place and the sample applications have been updated, actually reverted back to what the original design intended!

closed 2020-04-22 by JoelBender