JoelBender/bacpypes #308: Difference between reding properties from Yabe and real HAVC server

julia-tech, 2019-12-13:
Hi,

I would like to know that if there any difference between Yabe and real HVAC server? Right now I can read properties to Yabe with this code (https://github.com/JoelBender/bacpypes/blob/master/samples/HandsOnLab/Sample4_RandomAnalogValueObject.py), but I am not sure if I can directly apply this code in the real HVAC system? Is there anything I need to change or I can directly use the same code?

Thanks,
Julia

JoelBender, 2019-12-14:
Real HVAC systems have a wide variety of BACnet implementations, some of them support different combinations of properties and services while not supporting others.  There are two answers; (1) look on the Protocol Implementation Conformance Statement (PICS) that was provided by the manufacturer to be sure it supports the services you need, (2) read the protocol services supported property from the device object of the real HVAC system, (3) verify that it does with testing because the real proof is what happens "on the wire."  The code is the same.

And be careful of off-by-one errors :-).

ChristianTremblay, 2020-09-15:
No activity. Please reopen if required

closed 2020-09-15 by ChristianTremblay