JoelBender/bacpypes #309: How to read properties to WebCTRL (used for BACnet protocols) using BACpypes?

julia-tech, 2019-12-14:
Hi,

I use WHOIS-IAM to connect and then read properties to YABE. However, now I have a real HVAC system using WebController (WebCTRL http://www.automatedlogic.com/ ), can I still read properties using the same code? Or is there anything else I can do?

Thanks,
Julia

JoelBender, 2019-12-14:
Yes, it's the same code.  Note that a WebCTRL server doesn't have very much content that you can get via BACnet, it is a very simple BACnet server.  It is a very sophisticated client however, and you can learn a lot about BACnet by capturing traffic between the ALC server and the associated HVAC pieces of equipment.

julia-tech, 2019-12-15:
Hi, 

I have a question about using WebCTRL, how to use WHOIS-IAM to get the connection? Can I use https://github.com/JoelBender/bacpypes/blob/master/samples/HandsOnLab/Sample4_RandomAnalogValueObject.py directly without 'WHOIS-IAM' ?

Thanks,
Julia

JoelBender, 2019-12-15:
First, just a small point to terminology, BACnet is a connection-less protocol based on UDP and not TCP.  Once you know the address to communicate with another device you just sent it a request and get the response back, it's not like HTTP or other protocols where the request and response are inside a connection that provides a context for the exchange.

Every BACnet device has a unique identifier, created by a business process outside of the protocol, that is given to devices within a BACnet intranet.  If you are given the identifier, for example you are 1003 and you have been told a VAV box is device 4005, you can send out a broadcast **Who Is** looking for the device and it will respond with an **I Am**.  You then use the source of that message to "bind" to the device identifier, so every time you talk to the device like sending it a **Read Property** request you use that address.

If you know the address of the device, for example you are at 10.0.1.12 and someone told you that the VAV is at 10.0.1.75, you can send it a unicast **Who Is** without providing a device range and it will send you back an **I Am** with its device identifier along with other BACnet communications content.  This also completes the "binding" of device identifiers and addresses.

There is a lot that you don't *need* to know about the device to talk to it, you can simply send it **Read Property** requests for the value of **Analog Input Object 1** (written `analogInput:1` in the sample applications, stored as `('analogInput', 1)` in the Python code), but knowing that information really helps.

The sample application you've picked is a server, not a client, so it responds to whatever request is seen on the wire. It will respond to **Who Is** requests and **Read Property** requests that it has been given by a client, but will not initiate any itself.  If you would like to talk to the sample application from the WebCTRL server, you'll need to consult the server documentation on how to configure it to do that.

At some point during the configuration process I expect that the WebCTRL server will send out a **Who Is** request looking for your device running the sample, just like YABE would.

julia-tech, 2019-12-16:
Hi,

Thank you for your answer. So let me clarify that:

1. I still use the same code for 'WHOIS-IAM' to bind the connection first, right? 

Right now I set them in the same local web (e.g. 10.0.1.12 for WebCTRL (laptop #1) and 10.0.1.75 for another labtop #2 as client to run the Bacpypes code), then I successfully ping 10.0.1.12 (laptop #1) from laptop #2 (which means they are connected with the IP address). However, when I try to run 'WHOIS-IAM' in the client server (laptop #2), there is nothing shows when I type in 'WHOIS' or 'IAM' in the laptop #2. 

So my question is: does it mean the WebCTRL (as server) do not broadcast 'WHOIS' automatically? Or is there anything else I should do?

2. If I successfully build the connection using 'WHOIS-IAM', then I can directly use the 'Read Property' code, right? But if the WebCTRL has already connected with some real sensors (like temperature sensor etc.), do I need to change the 'AnalogValue: 1' to other name/number?

Thanks,
Julia

JoelBender, 2019-12-17:
In the configuration laptop #2 please confirm that you are using the CIDR notation for the address in the INI file, probably `10.0.1.75/24`.  It needs to know the size of the subnet to calculate the broadcast address for sending and receiving broadcast messages.  If you ran Wireshark (or the command line version called tshark) on the second laptop when the WebCTRL server starts up you'll see some interesting BACnet traffic on UDP port 47808.

Wireshark is an essential tool for protocol analysis because it is a neutral third party for presenting what is being sent on the wire.