JoelBender/bacpypes #311: How to add one more function to real HVAC system on WebCTRL?

julia-tech, 2019-12-18:
Hi,

Right now there are plenty of real HVAC devices connected to the WebCTRL, how can I add one more function such as (setpoint = 20) on the WebCTRL?

For example, previously the WebCTRL shows the temperature readings from the real device, now I need to add one more function like setpoint to that device on WebCTRL. Can I use Python to do that?

Thanks,
Julia

JoelBender, 2019-12-19:
I'm going to say no, but to be sure you need to check with your local dealer, and this is pretty obscure stuff, so you'll need to talk to ALC.

I'm saying no because while there are BACnet services that allow a client to create and delete objects they are not widely implemented.  Even if the HVAC device supports that service, there is no way to tell the device "this new object is a setpoint" for some value [1].  Everything that happens inside a device (how information is presented to applications, how the applications are structured, configured or programed) is a "local matter" which means it is up to the vendor to provide a tool to do that.  BACnet is there to help devices exchange data, and lots of metadata is available as well through optional properties if the vendor wants to take advantage of them.

[1] unless ALC tells you what properties and values a new object needs to do that

ChristianTremblay, 2020-09-15:
I don't think we need to keep this issue open. If further support needed, please reopen

closed 2020-09-15 by ChristianTremblay