JoelBender/bacpypes #312: How to issue a lighting command?

sedlacekdavid, 2019-12-22:
Hi, for first, thank you for the excellent work.

I have a lighting control object in my device, is it possible to control it over bacpypes? Is there any way to assemble a lighting command?

It's basically a Write Property mechanism I can use for controlling the Command property, if I understood it correctly, but now I just need to encode my command somehow.

Any help would be great, thank you, David

JoelBender, 2019-12-23:
The [312-lighting-command](https://github.com/JoelBender/bacpypes/tree/312-lighting-command) branch has a `WriteLightingCommand.py` that is a clone of the `ReadWriteProperty.py` sample code with the write command limited to Lighting Output objects and the property limited to the `lightingCommand` property.

It should be pretty clear what the code does, it treats the `LightingCommand` like any other kind of object and sets the appropriate values for the operation.  The order of the parameters is what is given in **Table 12-67**.  I have no way to test if this is working apart from being a well-formed request, please let me know!

sedlacekdavid, 2019-12-30:
Hi Joel! 
I have tested your code. And it works! Now I have to figure out how to use it without having to use the console.

Thank you for helping me, I've shot a short video on my phone if you are interested:

https://www.youtube.com/watch?v=UQnWa1BgAzQ

Keep it up!

Regards David

closed 2019-12-30 by sedlacekdavid