JoelBender/bacpypes #313: Python 3.8 Support

MichaelBMiner, 2020-01-08:
Hello,

 I was just wondering when will BACpypes support an install with pyton 3.8?

JoelBender, 2020-01-09:
I have heard of people that say it works fine on 3.8 now.  You can add a `(3, 7): 'py34',` line to the `setup.py` file if you would like to test it.

MichaelBMiner, 2020-01-09:
I have 3.8 installed and see no issues, but when trying to do a fresh install of bacpypes with 3.8, then bacpypes reports a failure.

JoelBender, 2020-01-09:
Maybe the `setup.py` application could check for a `--source-folder` option which bypasses the version check?  Meh.  The `bacpypes3` package will be one directory and Python3.6+ anyway.  I fixed the "add this line" comment above, may be that will be good enough for a while.

closed 2020-01-09 by MichaelBMiner