JoelBender/bacpypes #314: BinaryValue COV handling

noenthu, 2020-01-11:
Using bacpypes version 0.17.5 and 0.17.6 and I'm finding some odd behavior in the library parsing the received COV from a binary object. 

COV from some physical BACnet equipment when cast_out is shown as "active" or "inactive"
[image]

Using the Bacnet.Room.Simulator that is provided in the YABE repo, I get the following issue when trying to cast_out the COV for BinaryOutput, it is unable to get the value.
[image]

JoelBender, 2020-01-12:
This is a problem with their simulator as you have found.  The standard specifies that Binary Output Objects have a BinaryBV present value (active/inactive) but the simulator is providing a boolean (true/false).  Because the value is application tagged (that is to say it doesn't need a context to know how to interpret the tag data) you can turn it into a primitive datatype object and get the value:
```
>>> from bacpypes.primitivedata import Tag
>>> tag = Tag(Tag.applicationTagClass, Tag.booleanAppTag, 0, b'')
>>> tag.debug_contents()
    <bacpypes.primitivedata.Tag(boolean) instance at 0x7fe9f49af080>
        tagClass = 0 application
        tagNumber = 1 boolean
        tagLVT = 0
        tagData = ''
>>> tag.app_to_object()
<bacpypes.primitivedata.Boolean object at 0x7fe9f49af160>
>>> tag.app_to_object().value
False
```
There is way to "register" a subclass of BinaryOutputObject using the vendor identifier of the simulator and tell it that the present value is a boolean, but _(shortcoming of BACpypes follows)_ there is no way to tell the decoder that value.  There is an optional `vendor_id` parameter of `get_datatype()`, so if there _was_ a cache of I-Am contents, the address of the request/response could be looked up in that cache and  the vendor identifier extracted from that.  Anyway, it's something I've thought about :-).

noenthu, 2020-01-15:
Thanks @JoelBender that helps.
In the samples provided - ReadProperty for example, what is the recommended means by which I can convert the apdu.propertyValue to its primitive datatype object?

Would I need to read the tagClass, tagNumber, tagLVT from the dict_contents of the propertyValue and create a Tag?
It worked when I tried to cast_out, handled the InvalidTag exception, read the dict_contents and generated the Tag and read the value "False"

JoelBender, 2020-01-15:
The `apdu.propertyValue` is an instance of `Any` which is a wrapper around a `TagList`.  Inside that tag list is a tag.  So check that the tag list has exactly one application encoded tag and then use `app_to_object()`:
```python
property_value = apdu.propertyValue
assert len(property_value.tagList) == 1

property_value_tag = property_value.tagList[0]
assert property_value_tag.tagClass == Tag.applicationTagClass

value = property_value_tag.app_to_object().value
```

noenthu, 2020-01-15:
Thanks! That did the trick.

closed 2020-01-15 by noenthu