JoelBender/bacpypes #315: Using DeviceDiscovery.py with same IP address but different Device IDs and on the same network. Also, how to read the point values for those(as only the IP address is specified).

San42, 2020-01-15:
For example, 192.168.0.2 is the IP address for both the Devices 100 and 101. How do I read the point for device 101 and not 100? 
P.S I apologize if the the question has been answered before but I couldn't find it specifically.

noenthu, 2020-01-15:
@San42 are the devices running on different ports?

San42, 2020-01-15:
No, using the same port.

JoelBender, 2020-01-16:
BACnet/IP devices have unique addresses which are the combination of the IP address and the port.  So device 100 might be at `('192.168.0.2', 47808)` and device 101 at `('192.168.0.3', 47808)` using Python socket tuples.  If there are two applications using the same IP address then they _have_ to be using different ports or the operating system wouldn't let them run at the same time.

BACpypes uses a ":" to separate the IP address from the port, which defaults to 0xBAC0:
```
>>> from bacpypes.pdu import Address

>>> source_address = Address("192.168.0.2")
>>> source_address.addrTuple
('192.168.0.2', 47808)

>>> destination_address = Address("192.168.0.2:47809")
>>> destination_address.addrTuple
('192.168.0.2', 47809)
```
The program you are using can point the Who-Is request at a specific address, leaving out the high limit and low limit, for example, if you are  running on the default port:
```
> whois 192.168.0.2:47809
```
And as long as the device sends a _unicast_ response back (and most devices do these days) the program will continue doing it's thing.  Flipping that around, if you are running some other BACnet application that is already using the default port so you have to use a different one, the command is easier:
```
> whois 192.168.0.2
```