JoelBender/bacpypes #317: Not all COV Subscription notifications received if multiple triggered at near same time

noenthu, 2020-02-03:
I am running a modified COVServer.py from the example as a task with 250+ AnalogValue objects. Every 5-10 seconds, the presentValue of the objects is changed. The objects are stored in a global array.

Running a COV client, initial subscription returns all 250+ values as shown in log output, next notification shows fewer than 250, after a few iterations, the client only seems to log COV notifications from ~210 objects. 

The server shows log from the testanalogvalue task for all 250+ objects being changed.

Running python 3.6, Bacpypes 0.17.6

JoelBender, 2020-02-04:
I suspect this is a platform throughput issue.  There is no "back pressure" notification from the operating system back to the application, and UDP is by its nature an unconfirmed service (no guarantee of message delivery).

ChristianTremblay, 2020-09-15:
No activity. If you think it's a bug from bacpypes and not platform related, please reopen.

closed 2020-09-15 by ChristianTremblay