JoelBender/bacpypes #320: Managing Bacpypes thread from a different thread/python app

sedlacekdavid, 2020-02-12:
Hi, I've gone through the documentation and I'm still not sure about this. 

The goal is to be able to manage COV subscriptions on a Bacpypes thread in the runtime from a separate thread which runs it's own application. 

So I thought I'll use existing Bacpypes COV example, but I'm totally not sure how to trigger this subscription on the Bacpypes thread or anything else from a separate thread. I probably see how it can be done from Bacpypes console, but I don't need that console. There is a bunch of callback, which is wonderful, but I can't see a way how to pass any commands to be executed on a BAcpypes thread.

Can someone point me to the right direction?

thank you, David

JoelBender, 2020-02-13:
Is this a client-side application (you are managing subscriptions to objects on other devices) or a server-side application (you are managing the subscriptions of clients to your application)?

If you are a client, then you need to do your own thing about knowing when to initiate the COV subscription requests and how long they should be active, and of course knowing if the server device can support the volume of requests you might be making and the communications overhead.

If you are a server, then changing the value of one of the "properties tracked" should trigger the `send_cov_notifications()` method of the object (at least if the value changed enough according to the detection algorithm).  If you are not running in the BACpypes thread but some other thread, then you need to call `deferred()` which queues up your function call to be executed at the next opportunity in the BACpypes thread.

If you need to know when those notifications have been delivered and/or acknowledged, that's a harder problem.  If you look in the `ChangeOfValueServices.cov_notification()` function you'll notice that it creates an iocb but doesn't return it someplace where you can add a callback or wait for the completion event.

sedlacekdavid, 2020-02-13:
Hi Joel, it's purely a client app. I found examples how to issue a COV subscription, that's clear to me. But I don't know how to pass or generally manage Bacpypes application from another thread (same Python script). There is already another Python thread which manages other stuff and I want it to be in charge of subscribing Bacpypes thread to other devices. So that thread shall somehow communicate with Bacpypes thread. So the idea is to use Bacpypes thread as a sort of separation between my Bacnet devices and my Python application thread.
 
I can imagine I'll run BAcpypes with run(), but there is no API ( ?? ) to access the Bacpypes thread then? I can think of "hacking" it using the Console class and pass data/commands through a pipe to it, but that's just an ugly way and I still think there is a more appropriate solution.

JoelBender, 2020-02-13:
The BACpypes thread needs to keep "spinning along" in the `run()` loop, so to tell it you want something done call `deferred(fn, 1, x=3)`.  Inside the run loop you'll see another loop for calling deferred functions, so `fn(1, x=3)` will be called in the context of the BACpypes thread.

You can communicate back out of the BACpypes thread into your other threads by the usual signals, events, and queues.  For example, you could create an IOCB, pass that to your function with `deferred(fn, iocb)` which stuffs the results into `iocb.ioResult` and wait for the completion event.  Note that callbacks are not useful here because the callback function will be called in the _context of the BACpypes thread leading to snarls._

If you end up having lots of functions like this, you can wrap them in a controller like in the `Tutorial/ControllerAndIOCB.py` sample application, but I don't recommend it.  That is very old school, and the future is asyncio futures and tasks.

ChristianTremblay, 2020-02-13:
In BAC0, I usually define a "bacnet" object which is an BACpypes application by itself. Then I can interact with it from inside other threads.
One important thing though is to funnel all communication to 1 channel... So all read and write actions are defferred to the bacpypes thread. This way, I'm sure there is no collision or misbehaviours.

sedlacekdavid, 2020-02-16:
Joel, that's exactly what I needed. I thought deffered can be used only for the callbacks to be called after start of the bacpypes core/application.

Thank you, will try that.

closed 2020-02-16 by sedlacekdavid