JoelBender/bacpypes #322: ApplicationIOController not being called when mixing COV with IP2VLANRouter

bGridSven, 2020-03-12:
Here is the issue based on our conversation in the gitter chat. 
I am trying to make a gateway application which hosts multiple devices to which you can subscribe. However, whenever I try to subscribe to YABE the notication never gets send.

After investigation it seems that "the mix-in class list might not be working correctly."

```
#!/usr/bin/env python

"""
This sample application presents itself as a router sitting on an IP network
to a VLAN.  The VLAN has one or more devices on it with an analog
value object that returns a random value for the present value.

Note that the device instance number of the virtual device will be 100 times
the network number plus its address (net2 * 100 + n).
"""

import time
import random
import argparse

from threading import Thread

from bacpypes.debugging import bacpypes_debugging, ModuleLogger
from bacpypes.consolelogging import ArgumentParser

from bacpypes.core import run, deferred
from bacpypes.comm import bind

from bacpypes.pdu import Address, LocalBroadcast
from bacpypes.netservice import NetworkServiceAccessPoint, NetworkServiceElement
from bacpypes.bvllservice import BIPSimple, AnnexJCodec, UDPMultiplexer

from bacpypes.app import ApplicationIOController
from bacpypes.app import Application
from bacpypes.app import BIPSimpleApplication
from bacpypes.appservice import StateMachineAccessPoint, ApplicationServiceAccessPoint
from bacpypes.local.device import LocalDeviceObject
from bacpypes.local.object import CurrentPropertyList
from bacpypes.service.device import (
    WhoIsIAmServices,
    )
from bacpypes.service.object import (
    ReadWritePropertyServices,
    ReadWritePropertyMultipleServices,
    )

from bacpypes.service.cov import ChangeOfValueServices

from bacpypes.primitivedata import Real
from bacpypes.object import register_object_type, AnalogValueObject, Property

from bacpypes.vlan import Network, Node
from bacpypes.errors import ExecutionError

# some debugging
_debug = 1
_log = ModuleLogger(globals())

# globals
args = None

@bacpypes_debugging
class TestAnalogValueThread(Thread):

    """
    An instance of this class is created when '--avthread <interval>' is
    specified as a command line argument.  Every <interval> seconds it
    changes the value of the test_av present value.
    """
[... 198 lines cut ...]

        # make a random value object
        ravo = AnalogValueObject(
            objectIdentifier=('analogValue', 1),
            objectName='Temperature',
            presentValue=0.0,
            statusFlags=[0, 0, 0, 0],
            covIncrement=1.0,
            )

        vlan_app.add_object(ravo)

        test_av_thread = TestAnalogValueThread(ravo, 1)
        deferred(test_av_thread.start)

    _log.debug("running")

    run()

    _log.debug("fini")

if __name__ == "__main__":
    main()

```

JoelBender, 2020-03-12:
You are inheriting from the wrong classes when you make your stack.  The VLAN application is not sitting on a BACnet/IP network, it's on a VLAN, and if the other services aren't optional you can just inherit from them rather than adding the "capability" later:
```diff
@@ -102,6 +102,8 @@
 @bacpypes_debugging
 class VLANApplication(
-        BIPSimpleApplication,
+        ApplicationIOController,
         WhoIsIAmServices,
+        ReadWritePropertyMultipleServices,
+        ChangeOfValueServices,
         ):
```
And the initialization changes:
```diff
@@ -112,9 +114,5 @@
 
         # normal initialization
-        Application.__init__(self, vlan_device, aseID=aseID)
-
-        # optional read property multiple
-        self.add_capability(ReadWritePropertyMultipleServices)
-        self.add_capability(ChangeOfValueServices)
+        ApplicationIOController.__init__(self, vlan_device, aseID=aseID)
 
         # include a application decoder
```
And the debugging output of the APDUs being sent and received needs to be updated and this is the better way to do it:
```diff
@@ -147,17 +145,17 @@
     def request(self, apdu):
         if _debug: VLANApplication._debug("[%s]request %r", self.vlan_node.address, apdu)
-        Application.request(self, apdu)
+        super(VLANApplication, self).request(apdu)
 
     def indication(self, apdu):
         if _debug: VLANApplication._debug("[%s]indication %r", self.vlan_node.address, apdu)
-        Application.indication(self, apdu)
+        super(VLANApplication, self).indication(apdu)
 
     def response(self, apdu):
         if _debug: VLANApplication._debug("[%s]response %r", self.vlan_node.address, apdu)
-        Application.response(self, apdu)
+        super(VLANApplication, self).response(apdu)
 
     def confirmation(self, apdu):
         if _debug: VLANApplication._debug("[%s]confirmation %r", self.vlan_node.address, apdu)
-        Application.confirmation(self, apdu)
+        super(VLANApplication, self).confirmation(apdu)
 
 #
```

bGridSven, 2020-03-12:
Thank you very much, it is working now!

closed 2020-03-12 by JoelBender