JoelBender/bacpypes #328: BACpypes script producing no response then response

sagar448, 2020-03-27:
Hello Joel,

Firstly, amazing job on this library, really great work.

I've been adapting it to my use and so far so good, one problem that occurred was when I ran my custom script to read from a foreign device, for the first 10 responses it would print out No Response, but then after that it started printing out the actual response (e.g. some number) but then after a while it'll go back to no Response. Yes, I have adapted the script to run in a while loop. 

Do you know why this may be? I have attached an example.
[image]

Thanks for the help!

Kind Regards
Sagar Jaiswal

JoelBender, 2020-03-28:
Without capturing the actual traffic, it's very hard to say.  Reading from a foreign device shouldn't be any different than reading from any other IP device because the fact that it is foreign is at the virtual link layer, not the application layer.  However, a foreign device may consider itself "off line" and not respond to any requests unless it receives a registration acknowledgement from the BBMD it is registering to.  So maybe what you are seeing is the foreign device dropping offline until the TTL expires and it re-registers.

JoelBender, 2020-04-07:
There is a sample application **RecurringMultipleReadProperty.py** where you add the address, object identifier and property identifier to the `point_list` at the top of the application and let it run.  I would be curious if you get the same behavior.

sagar448, 2020-04-18:
Apologies for the late reply Joel. Thank you for your help! I took your advice ran the RecurringMultipleReadProperty...it produced the same problem but when I added a sleep of 5 seconds within the loop then 90% of the time it returns with data but 10% of the time it will still say noResponse. So I'm guessing it was because I'm pinging it too quick?

JoelBender, 2020-04-18:
If you can run Wireshark can capture all of the BACnet traffic (capture filter `udp port 47808`) and save that in a file, there are some scripts in the [bacpypes-pcap](https://github.com/JoelBender/bacpypes-pcap) project that might help summarize what's going on without going through each request/response one at a time.

ChristianTremblay, 2020-09-15:
No activity for a long time. If help still required, please reopen

closed 2020-09-15 by ChristianTremblay