JoelBender/bacpypes #329: Not able to view Raspberry Pi(running WhoIsIAm.py) data on WireShark(on PC)

shreyasrp-access, 2020-04-01:
I know this does not deal with the bacpypes library, but I wanted help with this as I am a noob.
I am able to see the whois and iam commands on WireShark when I enter it on the PC console after running WhoISIAm.py.
I then connected a Raspberry Pi 4 to the same WiFi as the PC, and entered the iam and whois commands on the Pi's console after running WhoIsIAm.py. But I am not able to see the data on WireShark. Are there some settings I need to change on my WiFi so that this works?

PS: The address in the INI file is edited according to my subnet mask and with CIDR notation.

JoelBender, 2020-04-01:
Try sending a unicast message, e.g, `whois 192.168.0.15` from the Pi to the PC and see if you can capture it.  If you can, there might be something about your broadcast settings of the access point because it should be acting like a normal switch.  If you are sending to the broadcast address for the network, like 192.168.0.255, then the Ethernet MAC layer should have a broadcast address so you should see something on the PC.  Then again, maybe the Pi isn't sending those as Ethernet broadcasts but something else.

Debug settings on the BACpypes applications won't help because this is in the socket library.  Maybe you can run tshark or daemonlogger on the Pi in the background and see what is appearing on the "wire."

shreyasrp-access, 2020-04-03:
[image]

I've made some development.
This is my setup, where the Desktop does not have WiFi and is connected to the Pi over ethernet. The Pi has 2 different IP addresses for its ethernet port (169.254.86.83) and wlan port (192.168.1.4).
I have WireShark on both my Laptop and Desktop.
When I change the address in the Pi's INI file to 169.254.86.83, I am able to see the iam packet on the Desktop's WireShark and not on Laptop. When the Pi's INI file has the address 192.168.1.4, I can see the packet on the Laptop's WireShark and not on Desktop.
Do you know if there is any way I can get the packets on both the laptop and desktop simultaneously?

JoelBender, 2020-04-03:
Ah, there's the ticket, you have two different interfaces.  When you run most BACnet software it binds to only one interface at a time, however routers bind to two or more.  There is an **IP2IPRouter.py** sample application which has two mini BACnet/IP stacks that are joined together at the the network layer.  Any _local broadcast_ by the PC will be seen by the Pi but not forwarded, any _global broadcast_ by the PC will be forwarded from one network to the other, adding some additional content in the NPCI (the networking piece of a BACnet packet), and vice versa.

So run the application on the Pi like this:
```bash
$ python IP2IPRouter.py 169.254.86.83/24 10 192.168.1.4/24 20
```
And you now have a BACnet intranet with two networks, 10 and 20.

_But how do I run **WhoIsIAm.py** at the same time?_

Ah, good followup question. The short answer is that you can't because you can't have two applications binding to the same port at the same time.  So you have to take those two applications and merge them together.  Let me know if you need help with that.

shreyasrp-access, 2020-04-04:
I tried it out and it works exactly you told. I could communicate between my Laptop and Desktop. Thank you very much for helping me out.
I was wondering if it is possible to merge the WhoIsIam.py and ReadAllProperties.py applications. When I run the program, I want it to automatically send a whois message, collect the different IP addresses and object IDs received from the devices, and run the ReadAllProperties on them to get their respective objects (binaryValue, analogValue,etc.).

For now, I edited the WhoIsIAm.py by removing the line:
`WhoIsIAmConsoleCmd()` 
and inserting
`addr = GlobalBroadcast()
this_application.who_is(None, None, addr)`
Running this will automatically print the device data and then gets stuck.

JoelBender, 2020-04-04:
That function will send out the message, but there's nothing that tells it when to stop.  The Who-Is-Request service request is an unconfirmed request, there is no way to know if a device will respond or how many devices are supposed to respond.  That comment in the code `put the parameters someplace where they can be matched when the appropriate I-Am comes in` is left up to "upper layers" of the application.

The program you are looking for is `samples/DiscoverToDo.py` which puts all the stuff it finds into a tab-delimited text file.  Each row of the file has (devid, objid, propid, version, value) where version is an incrementing integer if the value changes between reads (so you can do a "diff").

The "ToDo" parts of the program are for keeping track of what needs to happen and what to do with the response when it happens, conceptually like a thread without using threads.

shreyasrp-access, 2020-04-06:
I cannot find the DiscoverToDo.py file in the samples folder.
I ran the WhoIsIAm.py, entered whois, and got data from the RaspberryPi (which is running mini_device.py), so the communication is happening.
Then I tried running the Discover.py file as `python Discover.py stuff` 
`>whois`
It did not print anything on the terminal, although I saw the i-am response on WireShark.

`>EOF`
It created a file called stuff, but that file is empty.

The WhoIsIAm.py I was talking about is the one in samples/Tutorial.
The WhoIsIAm.py in samples also does not print the response for whois in the terminal, but the response shows up in WireShark.

JoelBender, 2020-04-07:
Ah!  You are correct, the **DiscoverToDo.py** is in the `stage` branch, I'll ix that before the next release.

Please carefully check the subnet mask of your device.  If you left off the `/24` or whatever is appropriate for you LAN then the Who-Is could be going out to all devices but it won't hear the response.

shreyasrp-access, 2020-04-08:
The subnet mask has to be right (else `samples/Tutorial/WhoIsIAm.py` would not work).

I found out that the do_whois() function of the class WhoIsIAmConsoleCmd is different for `samples/Tutorial/WhoIsIAm.py` and `samples/WhoIsIAm.py`.
`samples/Tutorial/WhoIsIAm.py` works fine. There must be a bug in `samples/WhoIsIAm.py` and `samples/Discover.py`.

I have some other questions too:
1. Is it possible to get the MAC address of a BACnet device using bacpypes?
2. Do BACnet/IP devices need a separate network? Or is it possible for them to co-exist with other IP devices in the same LAN? Will there be an issue with traffic when broadcasts are sent?
3. When the BACnet/IP router is used with  BACnet MSTP devices, will the devices send the response in the same format as it is when we communicate directly with BACnet IP devices?

JoelBender, 2020-04-08:
Ah, yes, there's is a balance between having lots of little applications that do specialized things and monsters.  Not that it wouldn't also be nice to say `$ python -m bacpypes` and have a shell.

1) No, the MAC address is not available in the PDU from the socket library.  In theory you can open a raw socket and filter the things you want, but I would recommend using libpcap (there is some same decoding in the `bacpypes.analysis` module that might help)

2) No, BACnet/IP devices can live on the same network with everything else.  BACnet/IP uses an IANA assigned port, and while there's a chance another application might step on the traffic, it's fine in the real world.

3) The application layer is the same (which is the fixed encoded part called the APCI and the variable encoded part) but the network layer is different.  When the packet goes from the MS/TP device to the router is as the DNET/DLEN/DADR fields set with the BACnet network number and IPv4 address.  When the router forwards the message to the IP device, the DNET/DLEN/DADR is removed and the SNET/SLEN/SADR fields are added.  See **Clause 6**.

shreyasrp-access, 2020-04-15:
[image]

Here, the laptop and router are in the same network. The laptop sends requests to the router which 
 forwards them to the devices. (The laptop will run commands only using bacpypes)
1. If I send a whois command, will I get only the router's response, but also of all the MSTP devices?
2. When the MSTP network is assigned a network number of 20, does it bind with the IP address of the router? When the laptop tries to communicate to network 20, how will it know to choose this specific router without mentioning its IP address?
3. In the above image, suppose I have a gateway (which converts MODBUS to BACnet IP) instead of the router and MODBUS devices instead of BACnet devices, will the commands in bacpypes still work? If I want to run `ReadAllProperties.py`, what are the arguments to be passed?

JoelBender, 2020-04-17:
1) There are a few forms of Who-Is messages, the simplest are local broadcast to reach all of the devices that are on the "local" network, and a global broadcast which is cascaded throughout the entire network.  If you way `whois *` then you'll get local ones and `whois *:*` you'll send it everywhere.

2) The idea of a network is in the mind of the router.  Assuming the simple case of two connections, there will be a BACnet network number for each one.  When a router starts up it send a local broadcast I-Am-Router-To-Network 10 on the connection to 20, telling all of the devices on twenty to send to it to forward traffic.  It likewise sends an I-Am-Router-To-Network 20 to the interface for 10.  If the application layer message says "send this to address X on network 20" and it doesn't know the router, it first sends out a Who-Is-Router-To-Network 20 and the router responds.  Then the message is directed to the router with a header that says "please forward this on for me."

3) That depends on how well the gateway is providing a virtual BACnet device implementation and how it maps objects.  In the simplest case, the gateway will have one virtual BACnet device for each MODBUS device which will then have one virtual Analog Value Object for each holding register, one Binary Value Object for each coil, etc.  Reading the present value of the AVO could directly translate it to a MODBUS function, or there could be a "scan" routine that keeps a local cache of MODBUS values and the Read Property request is served out of the cache (I don't have as many examples for [MODpypes](https://github.com/JoelBender/modpypes) applications).

ChristianTremblay, 2020-09-15:
New tag for future reference. 
No activity. Please reopen if further assistance is needed.

closed 2020-09-15 by ChristianTremblay