JoelBender/bacpypes #330: LocalScheduleObject and changing exceptionSchedule

kazemiprince, 2020-04-05:
Hello and I wish all of you are ok.
I'm trying to create a LocalScheduleObject in order to schedule another local objects. (application is run on raspberrypi)
every thing is working fine, but when I write the exceptionSchedule property from another device (my test is from my laptop) new schedule will not apply until restarting the application. 
I can see exceptionSchedule  changed, but I have to restart the application to take action on binded objects and presentValue property as well.
So I decided to restart the application on case of write property, but do to restriction on Threads, I can't find a effective way to terminate run() and using the Process instead of Thread make lots of none synced variables on bacpypes process and main.
please advice how to apply new schedule without restarting the whole application.

JoelBender, 2020-04-05:
Please run your test again and turn on debugging like `--debug bacpypes.local.schedule:snork.txt` and upload the text file to the issue.  My guess is that the `LocalScheduleInterpreter.process_task()` isn't being called after the exception schedule has changed.

kazemiprince, 2020-04-05:
thanks for your kind attention.
attached is the logs
[first.txt](https://github.com/JoelBender/bacpypes/files/4434438/first.txt)
[second.txt](https://github.com/JoelBender/bacpypes/files/4434439/second.txt)

first.txt file is including write from another device, that schedule not update
second.txt log is after restarting application

kazemiprince, 2020-04-05:
thanks for your support
adding 

`obj._task.process_task() `

in write property solved my problem.

JoelBender, 2020-04-05:
Well, that's good, but it should do that automatically!  Notice that there are lines like this:
```
DEBUG:bacpypes.local.schedule.LocalScheduleInterpreter:present_value_changed Real(20) Real(55)
DEBUG:bacpypes.local.schedule.LocalScheduleInterpreter:    - error: ExecutionError('property', 'writeAccessDenied')
```
This is a separate problem.  The object that you are writing to (in the list of object property references) is not writable.  Subclass it and provide a writable property similar to this:
```
@register_object_type(vendor_id=999)
class WritableAnalogValueObject(AnalogValueObject):
    properties = [WritableProperty("presentValue", Real)]
```

kazemiprince, 2020-04-05:
thanks for your hint, today debug show my mistake in copy and paste! 
below line was at end of all of write properties that I don't know what was the if.. in the begining that I removed before.. ..
`raise ExecutionError(errorClass="property", errorCode="writeAccessDenied")`

anyway, that one solved too.
here is whole code of analogValue object
`
@bacpypes_debugging
class AVPresentValue(WritableProperty):
    def __init__(self, identifier):
        if _debug:
            AVPresentValue._debug("__init__ %r", identifier)
        WritableProperty.__init__(
            self, identifier, Real, default=0.0, optional=False, mutable=False
        )

    def ReadProperty(self, obj, arrayIndex=None):
        if _debug:
            AVPresentValue._debug("ReadProperty %r arrayIndex=%r", obj, arrayIndex)

        # access an array
        if arrayIndex is not None:
            raise ExecutionError(
                errorClass="property", errorCode="propertyIsNotAnArray"
            )

        
        value = obj._AV.value

        return value

    def WriteProperty(self, obj, value, arrayIndex=None, priority=None, direct=False):
        if _debug:
            AOPresentValue._debug(
                "WriteProperty %r %r arrayIndex=%r priority=%r direct=%r",
                obj,
                value,
                arrayIndex,
                priority,
                direct,
            )
        obj._AV.value = value

        #raise ExecutionError(errorClass="property", errorCode="writeAccessDenied")

#
#   analogValue BACPYPES
#

@bacpypes_debugging
@register_object_type
class RPiAnalogValue(AnalogValueObject):

    properties = [AVPresentValue("presentValue")]

    def __init__(self, AV_object, AV_id, **kwargs):
        if _debug:
            RPiAnalogValue._debug("__init__ %r %r", AV_id, kwargs)
        AnalogValueObject.__init__(self, **kwargs)
        self._AV = AV_object
       `

kazemiprince, 2020-04-05:
it was my mistake in copy and paste of codes.
below is whole of codes for analogValue object.
`

@bacpypes_debugging
class AVPresentValue(WritableProperty):
    def __init__(self, identifier):
        if _debug:
            AVPresentValue._debug("__init__ %r", identifier)
        WritableProperty.__init__(
            self, identifier, Real, default=0.0, optional=False, mutable=False
        )

    def ReadProperty(self, obj, arrayIndex=None):
        if _debug:
            AVPresentValue._debug("ReadProperty %r arrayIndex=%r", obj, arrayIndex)

        # access an array
        if arrayIndex is not None:
            raise ExecutionError(
                errorClass="property", errorCode="propertyIsNotAnArray"
            )

        
        value = obj._AV.value

        return value

    def WriteProperty(self, obj, value, arrayIndex=None, priority=None, direct=False):
        if _debug:
            AOPresentValue._debug(
                "WriteProperty %r %r arrayIndex=%r priority=%r direct=%r",
                obj,
                value,
                arrayIndex,
                priority,
                direct,
            )
        obj._AV.value = value

        #raise ExecutionError(errorClass="property", errorCode="writeAccessDenied")

#
#   analogValue BACPYPES
#

@bacpypes_debugging
@register_object_type
class RPiAnalogValue(AnalogValueObject):

    properties = [AVPresentValue("presentValue")]

    def __init__(self, AV_object, AV_id, **kwargs):
        if _debug:
            RPiAnalogValue._debug("__init__ %r %r", AV_id, kwargs)
        AnalogValueObject.__init__(self, **kwargs)
        self._AV = AV_object
       
`

at last, thank you very much for your kind attention

closed 2020-04-05 by kazemiprince

JoelBender, 2020-04-07:
Fixed in `stage` branch for next release.