JoelBender/bacpypes #331: writeProperty on local device

kazemiprince, 2020-04-06:
Hello everyone.
I'm looking for a way to do writeProperty on local device but I could not find.
local device is  BIPSimpleApplication and I wish ti write from another part of app (GUI) to the localSchedule object.

bacnet is run one another Thread.
I used below function but it replied with noResponse, even it's fine for write to other devices not local.

`   

    def do_write(self, addr, obj_id, prop_id , value):
             """write <addr> <type> <inst> <prop> <value> [ <indx> ] [ <priority> ]"""
              #args = args.split()
              #ReadWritePropertyConsoleCmd._debug("do_write %r", args)

        try:
            #addr, obj_id, prop_id = args[:3]
            obj_id = ObjectIdentifier(obj_id).value
            #value = args[3]

            indx = None
            # if len(args) >= 5:
            #     if args[4] != "-":
            #         indx = int(args[4])
            # if _debug: ReadWritePropertyConsoleCmd._debug("    - indx: %r", indx)

            priority = None
            # if len(args) >= 6:
            #     priority = int(args[5])
            # if _debug: ReadWritePropertyConsoleCmd._debug("    - priority: %r", priority)

            # get the datatype
            datatype = get_datatype(obj_id[0], prop_id)
            #if _debug: ReadWritePropertyConsoleCmd._debug("    - datatype: %r", datatype)

            # change atomic values into something encodeable, null is a special case
            if (value == 'null'):
                value = Null()
            elif issubclass(datatype, AnyAtomic):
                dtype, dvalue = value.split(':', 1)
                #if _debug: ReadWritePropertyConsoleCmd._debug("    - dtype, dvalue: %r, %r", dtype, dvalue)

                datatype = {
                    'b': Boolean,
                    'u': lambda x: Unsigned(int(x)),
                    'i': lambda x: Integer(int(x)),
                    'r': lambda x: Real(float(x)),
                    'd': lambda x: Double(float(x)),
                    'o': OctetString,
                    'c': CharacterString,
                    'bs': BitString,
                    'date': Date,
                    'time': Time,
                    'id': ObjectIdentifier,
                    }[dtype]
                #if _debug: ReadWritePropertyConsoleCmd._debug("    - datatype: %r", datatype)

                value = datatype(dvalue)
                #if _debug: ReadWritePropertyConsoleCmd._debug("    - value: %r", value)

            elif issubclass(datatype, Atomic):
                if datatype is Integer:
                    value = int(value)
                elif datatype is Real:
                    value = float(value)
                elif datatype is Unsigned:
                    value = int(value)
                value = datatype(value)
            elif issubclass(datatype, Array) and (indx is not None):
                if indx == 0:
                    value = Integer(value)
                elif issubclass(datatype.subtype, Atomic):
                    value = datatype.subtype(value)
                elif not isinstance(value, datatype.subtype):
                    raise TypeError("invalid result datatype, expecting %s" % (datatype.subtype.__name__,))
            elif not isinstance(value, datatype):
                raise TypeError("invalid result datatype, expecting %s" % (datatype.__name__,))
            #if _debug: ReadWritePropertyConsoleCmd._debug("    - encodeable value: %r %s", value, type(value))

            # build a request
            request = WritePropertyRequest(
                objectIdentifier=obj_id,
                propertyIdentifier=prop_id
                )
            request.pduDestination = Address(addr)

            # save the value
            request.propertyValue = Any()
            try:
                request.propertyValue.cast_in(value)
            except Exception as error:
                print("WriteProperty cast error: %r", error)

            # optional array index
            if indx is not None:
                request.propertyArrayIndex = indx

            # optional priority
            if priority is not None:
                request.priority = priority

            #if _debug: ReadWritePropertyConsoleCmd._debug("    - request: %r", request)

            # make an IOCB
            iocb = IOCB(request)
            #if _debug: ReadWritePropertyConsoleCmd._debug("    - iocb: %r", iocb)

            # give it to the application
            self.this_application.request_io(iocb)

            # wait for it to complete
            iocb.wait()

            # do something for success
            if iocb.ioResponse:
                # should be an ack
                if not isinstance(iocb.ioResponse, SimpleAckPDU):
                    print("    - not an ack")
                    return

                print("ack\n")

            # do something for error/reject/abort
            if iocb.ioError:
                print(str(iocb.ioError) + '\n')

        except Exception as error:
            print("exception: %r", error)`

JoelBender, 2020-04-07:
If this is all in the same process, it's much simpler.  Assuming that you have a reference `some_av` that is an Analog Value Object, you can change primitive values directly with `some_av.presentValue = 75.3`.  If there are more complicated changes, you can call something like `some_av.WriteProperty('presentValue', 75.3, priority=5)`.

If the objects are added/removed to the application (I often use `this_application` in the sample code) then a dictionary of objects by name is in `this_application.objectName` and by object identifier in `this_application.objectIdentifier`.

kazemiprince, 2020-04-07:
Thanks Master!
solved.
but I didn't find these in instructions! Am I blind or instructions are not complete?

JoelBender, 2020-04-07:
You are not blind, padawan, these gems are carefully hidden from view to make the process of discovery a much more satisfying thrill.  You must use the force and read the source!  The master is far to busy burying new gems to be distracted by trying to keep the ancient scrolls current.

:-)

closed 2020-04-07 by JoelBender