JoelBender/bacpypes #332: Not able to read BACnet MSTP points connected to BACnet IP Router

shrutifiske, 2020-04-07:
Hi,

I am successfully able to get points that are connected to the BACnet IP router in the field. But I am not able to get some digital and analog inputs/outputs points that are connected to BACnet MSTP which is then connected to the BACnet IP router. Where I am getting these MSTP points in the BACnet Discovery tool but not using bacpypes. Can you please help me here.

JoelBender, 2020-04-07:
Are you in the same IP broadcast domain (on the same subnet) as the router?  I suspect that the **Who-Is-Router-To-Network** message that is being sent isn't getting received by the router.  If you run the application with the `--route-aware` option you can send a Who-Is with a global broadcast directly at the router, e.g., `whois *:*@192.168.2.15`.  There is a chance that the devices broadcast their **I-Am** responses so you won't get them, but recommendation is for **I-Am** messages to be unicast back and most devices follow that.

ChristianTremblay, 2020-09-15:
Probable network configuration issue. Closing. If help still needed, please reopen

closed 2020-09-15 by ChristianTremblay