JoelBender/bacpypes #342: Issuing COV subscriptions

sedlacekdavid, 2020-06-15:
Hi Guys. Have a question regarding COV subscriptions. 

Currently I'm issuing COV subs using deffered(). Could you please check if I'm doing it correctly if I'm doing following. 
I iterate over a set of SubscribeCOVRequest() subs. Like this:
```
        # make an IOCB
        iocb = IOCB(request)

        # callback when it is acknowledged
        iocb.add_callback(self.subscription_acknowledged)

        # give it to the application
        self.request_io(iocb)
```

I can see in the log that they are being processed:
```send_subscription <COVClientApp.SubscriptionContext object at 0x04799B50>
    - iocb: <bacpypes.iocb.IOCB(2) instance at 0x047a8a90>
send_subscription <COVClientApp.SubscriptionContext object at 0x04799C50>
    - iocb: <bacpypes.iocb.IOCB(3) instance at 0x04f465b0>
send_subscription <COVClientApp.SubscriptionContext object at 0x04799C10>
    - iocb: <bacpypes.iocb.IOCB(4) instance at 0x04f46950>
send_subscription <COVClientApp.SubscriptionContext object at 0x04799D10>
    - iocb: <bacpypes.iocb.IOCB(5) instance at 0x04f468f0>
send_subscription <COVClientApp.SubscriptionContext object at 0x04799D50>
    - iocb: <bacpypes.iocb.IOCB(6) instance at 0x04f46bf0>
send_subscription <COVClientApp.SubscriptionContext object at 0x04799DB0>
    - iocb: <bacpypes.iocb.IOCB(7) instance at 0x04f46c90>
send_subscription <COVClientApp.SubscriptionContext object at 0x04799E30>
    - iocb: <bacpypes.iocb.IOCB(8) instance at 0x04f46d90>

subscription_acknowledged <bacpypes.iocb.IOCB(2) instance at 0x047a8a90>
    - error: <bacpypes.apdu.AbortPDU(None,1) instance at 0x4f556d0>
subscription_acknowledged <bacpypes.iocb.IOCB(3) instance at 0x04f465b0>
    - error: <bacpypes.apdu.AbortPDU(None,2) instance at 0x4f46210>
subscription_acknowledged <bacpypes.iocb.IOCB(4) instance at 0x04f46950>
    - error: <bacpypes.apdu.AbortPDU(None,3) instance at 0x4f46430>
subscription_acknowledged <bacpypes.iocb.IOCB(5) instance at 0x04f468f0>
    - error: <bacpypes.apdu.AbortPDU(None,4) instance at 0x4f46550>
subscription_acknowledged <bacpypes.iocb.IOCB(6) instance at 0x04f46bf0>
    - error: <bacpypes.apdu.AbortPDU(None,5) instance at 0x4f46870>
subscription_acknowledged <bacpypes.iocb.IOCB(7) instance at 0x04f46c90>
    - error: <bacpypes.apdu.AbortPDU(None,6) instance at 0x4f46970>
subscription_acknowledged <bacpypes.iocb.IOCB(8) instance at 0x04f46d90>
    - error: <bacpypes.apdu.AbortPDU(None,7) instance at 0x4f465b0>
subscription_acknowledged <bacpypes.iocb.IOCB(9) instance at 0x04f46df0>
    - error: <bacpypes.apdu.AbortPDU(None,8) instance at 0x4f46b70> 
```

BUT - in wireshark, I can see how the subs being propagated IMMEDIATELY to the network, and can see there are immediate confirmations being returned back. 

But from some reason, bacpypes doesn't pick up those confirmations and instead of that it starts slowly printing one by one 
```subscription_acknowledged <bacpypes.iocb.IOCB(2) instance at 0x047a8a90>
    - error: <bacpypes.apdu.AbortPDU(None,1) instance at 0x4f556d0>
..............
```
in ~ 3 second intervals. 

Could you please check what's going on?

OR - should I issue a subscribe and then WAIT until I get a callback and then continue with the next sub and so on?

Thank you very much, 
David

sedlacekdavid, 2020-06-15:
It seems like bacpypes doesn't pick up any response coming back.

sedlacekdavid, 2020-06-15:
Isn't it a problem that the IDs below are not matching??

```
subscription_acknowledged <bacpypes.iocb.IOCB(**4)** instance at 0x04f46950>
    - error: <bacpypes.apdu.AbortPDU(None,**3**) instance at 0x4f46430>
```

JoelBender, 2020-06-15:
Is this a multithreaded application?  Are you calling request_io in the same thread as a the `run()` function?  Please change this:
```
        # give it to the application
        self.request_io(iocb)
```
to this:
```
        # give it to the application
        deferred(self.request_io, iocb)
```
The two numbers you have highlighted are different, the IOCB is an internal identifier, for the AbortPDU the number is the invoke id.  The 'None' part is confusing.

sedlacekdavid, 2020-06-16:
@JoelBender it runs on it's own thread. 

I've tried what you suggested, but with no luck...

I've relaized one thing - I successfuly start the threads, but I'm calling deffered() (in subscribeTo() body below) from the other thread than Bacpypes is running on. 
Like this:
```
@staticmethod
    def subscribeTo(_address: str, _callback, _map: Map):
        context = SubscriptionContext(Address(_address), _map, _callback, True, 60)
        deferred(BacpypesThread.this_application.send_subscription,
                 context)
```

But I guess that's not an issue ?

sedlacekdavid, 2020-06-16:
I can see how deffered() successfully fills and being executed.

[image]

closed 2020-06-16 by sedlacekdavid