JoelBender/bacpypes #343: Address class should support initialization with ip_address class

ottobackwards, 2020-06-23:
It would be nice if you could just pass in the ip_address instance and not have to Address(str(address))

JoelBender, 2020-06-23:
Are you referring to instances of objects returned by [ipaddress.ip_address()](https://docs.python.org/3/library/ipaddress.html#ipaddress.ip_address)?

ottobackwards, 2020-06-24:
yes

ottobackwards, 2020-06-24:
Do you accept PRs?

ottobackwards, 2020-06-24:
although, I guess you would not want a different version dependent parameter or tests?  you can close

JoelBender, 2020-06-25:
The ipaddress module has been [back ported to 2.7](https://pypi.org/project/ipaddress/) and is builtin for 3.5+, so in theory the feature could be conditionally supported the way netifaces is now.  The downside is that it specifically requires unicode strings so the ability to use `Address(ipaddress.ip_address("1.2.3.4"))` the way you would think it _should_ work wont.  Supporting in only the py34 source directory is still an option, so if you'd like to submit a PR that would be fine (and tests would be icing).

The good news is that bacpypes3 supports both IPv4 and IPv6 addresses out of the box and `bacpypes3.pdu.IPv4Address` is a subclass of `ipaddress.IPv4Interface`.

ottobackwards, 2020-06-25:
bacpypes3?  what is this?

ottobackwards, 2020-06-25:
do you have any docs on building and running the tests for PR submittals etc?

JoelBender, 2020-06-25:
bacpypes3 is a version of BACpypes for Python3.6+ that is based on asyncio and async/await; the core is replaced by the event loop, iocb's are replaced with futures/tasks, etc.  It will support IPv4 and IPv6, and SC using the [websockets](https://websockets.readthedocs.io/en/stable/) library.

The tests are in the tests directory based on **pytest** and you should be able to run them with a `pythonX.X setup.py test` and that's it.  I have dipped my toes in the **tox** waters to run the tests on multiple Python versions, which seems fine when they all work, but I drop back to clones the `test_script.sh` to focus on individual tests when they fail.  The tests are the same for all of the source versions, so at some point you'll have to deal with the `u"1.2.3.4"` madness, and I've given up on the Python2.5 tree so it really should be deleted.

ottobackwards, 2020-06-26:
In the end, a hasty issue comes to a hasty end.   The gain by using ipaddress ( because I was using it as argparse type for parameter handling ) doesn't justify the changes to handle it.

Thanks for the time

closed 2020-06-26 by ottobackwards