JoelBender/bacpypes #348: Is there a way to get only the Mac Address from a who is request?

shuffguy, 2020-07-25:
Apologies for such a trivial question but I couldn't find an answer through googling or the documentation. Is there a way I can determine the mac address, and only the mac address, of the device? So when I do a whois request I get

`('device', 10001) is at 192.168.15.207:47809 named 'BASRT-PT01'`

But this address seems hardcoded together. Is there a way I can parse the address object to only get 47809, which I assume is the mac address?

JoelBender, 2020-07-26:
The 47809 is the UDP port number that is the source of the packet, in the Python socket library it would be the second half of the tuple ('192.168.15.207', 47809).  The actual MAC address, assuming that IPv4 is running on an Ethernet network, is hidden by the operating system in the Address Resolution Protocol (ARP) tables.  In Linux you can run the `arp` command and see the contents of the table if the IPv4 address is on your network (if it has been routed to get to you, that information doesn't exist on your network).

The BACnet/IPv4 "virtual link layer" treats these as "virtual MAC addresses".  But besides that, if you have an `Address` object that is a IPv4 address you can get the tuple used in the socket layer like this:
```
>>> addr = Address("192.168.15.207:47809")
>>> addr.addrTuple
('192.168.15.207', 47809)
```
If this packet went through a BACnet router to get to you, the source address will have the network number in front so it will look like this:
```
>>> addr = Address('15:192.168.15.207:47809')
>>> addr.addrNet
15
>>> addr.addrTuple
('192.168.15.207', 47809)
```
Note that the tuple is still the IPv4 address and port number, and the `addrNet` has the BACnet network number.

ChristianTremblay, 2020-09-15:
And you will need to make a read request on that device to fetch the object name. It won't come with a simple WHOIS.