JoelBender/bacpypes #351: Need help to know the format of the DataTimeValueObject 'Present Value'(DateTime) format which bacpypes accept

snehabati, 2020-08-10:
Hi,
I am new to bacpypes and python as well, hence need some help in knowing the format of the  
DataTimeValueObject 'Present Value' which is DateTime datatype. I am trying to mock a device using bacpyes as an instance for which I need to put in the valid present value format of DateTime datatype.

I tried to get it this way put seems like it doesn't work, can you please help me.

from bacpypes import primitivedata
from bacpypes import basetypes

D=primitivedata.Date(year=116,month=11,day=28,day_of_week=1)
print("Date=",D)
T=primitivedata.Time(hour=0,minute=0,second=0,hundredth=0)
print("Time=",T)
PresVal= presval_datetimeval= basetypes.DateTime(date=D,time=T)
print("Present_Value of DateTimeValue Object= ",PresVal)

Output:

Date= Date(2016-11-28 mon)
Time= Time(00:00:00.00)
Present_Value of DateTimeValue Object=  <bacpypes.basetypes.DateTime object at 0x00000190773C9748>

Process finished with exit code 0

JoelBender, 2020-08-11:
Check out Clause 20.2.12 Encoding of a Date Value:
> Date values shall be encoded in the contents octets as four binary integers. The first contents octet shall represent the year minus 1900; the second octet shall represent the month, with January = 1; the third octet shall represent the day of the month; and the fourth octet shall represent the day of the week, with Monday = 1. A value of X'FF' = D'255' in any of the four octets shall indicate that the corresponding value is unspecified and shall be considered a wildcard when matching dates. If all four octets = X'FF', the corresponding date may be interpreted as "any" or "don't care."

The value of a BACpypes primitive Date object is a 4-tuple of exactly these values.  The class provides `__str__()` and `__repr__()` functions to make them easier to understand, which you have already discovered.  The DateTime object is a "sequence", which in ASN.1 terms is like a structure, with individual named components, so you `DateTime(date=D, time=T)` is exactly correct.  Unlike the primitive types, there is no special `__str__()` methods for printing these things, but there is a `debug_contents()` function which is similar:
```
>>> PresVal.debug_contents()
    date = <bacpypes.primitivedata.Date object at 0x7f1f1d8a2fd0>
    time = <bacpypes.primitivedata.Time object at 0x7f1f1d8a9080>
```
It doesn't use the `__str__()` function for displaying primitive data values, perhaps it should.  But the real values are still in there, tucked away:
```
>>> PresVal.date
<bacpypes.primitivedata.Date object at 0x7f1f1d8a2fd0>
>>> PresVal.date.value
(116, 11, 28, 1)
```
You can try different combinations of the 4-tuple:
``` you should not be too

snehabati, 2020-08-12:
ok thanks much Joel, that helped. My issue is solved.

closed 2020-08-12 by JoelBender