JoelBender/bacpypes #353: Any method defined in Bacpypes to get Device's All Objects Alarm Summary

snehabati, 2020-08-18:
Hi,

Is there any method in bacpypes to fetch all objects Alarm Summary?
Suppose for example in my Device which has Binary_Input Object and it has alarm active, similary in I have another instance on Binary_Input having another alarm and also another object which has alarm active, I want to to know that these there have alarms active. Is there any method which goes and checks the alarmValue of all the objects and gives a summary kind?

Thanks,
Sneha

JoelBender, 2020-08-18:
No, but it shouldn't be too hard to scan though the application values of the `objectName` _or_ the `objectIdentifier` of an application (the two dictionary values should be the same).  Add a "help wanted" label if you would like help writing a `do_GetAlarmSummary()` function as a mix-in class for an application so it will return a GetAlarmSummaryACK.

snehabati, 2020-08-19:
Yes please need help for the function.
Also I would like to seek a clarification on the Alarm Detection, I have set the object property 'Alarm Value' as 1 which means active in my mock device. Should this property setting be enough to produce the Alarm?

JoelBender, 2020-08-19:
There are a few properties that the service depends on, but before that, I noticed that the **GetAlarmSummary** service is deprecated, first paragraph Clause 13.10:

> The specification of this service is retained for historical reference so that implementations of client devices have guidance on
how to interoperate with older server devices. Otherwise, it is deprecated. It is recommended that execution of this service
not be implemented in server devices.

More specific to your question for this service, the server will "search all (of its) event-initiating objects that have an
Event_State property not equal to NORMAL and a Notify_Type property whose value is ALARM. Any object that has an
Event_Detection_Enable property with a value of FALSE shall be ignored."

Note also that **GetEnrollmentSummary** (Clause 13.11) service is also deprecated in favor of **GetEventInformation** (Clause 13.12).  So we'll need to dig into this a little deeper.

snehabati, 2020-08-19:
The explanation on the device object properties value that raises the alarm is very clear, also for the services deprecated info, Thanks much.
One more question I have in mind, as AlarmSummary and few other services are deprecated now, are there any other functions  which I can use so that my client can be notified of the alarms?

JoelBender, 2020-08-19:
Clause 13, Alarm and Event Services, is the chapter.  Note that "... BACnet makes no distinction between an alarm and an event."  You can do all kinds of things on the server side to help your clients, but if you have a specific client in mind then you should find out what it supports and save yourself a lot of work.

snehabati, 2020-08-20:
ok, thank you for the info. I will create function for fetching the device objects in alarm status.
Before writing the client side function I wanted to see how yabe explorer(client) responds if the device object has alarm.
I tried to set my mock device's(bacpypes instance) AV object in alarm status by setting its property values  as below:
"eventState": 3 , "statusFlags": [0], "notifyType": 0,"eventDetectionEnable": True
Then I used Yabe explorer to see the alarm status of the object in logs, I could not see any alarm related logs in yabe.(PFA screenshot for reference) 
[image]

Will setting of the above properties notify the client(Yabe) of the object in alarm state by default or do I need to do some more things to see the alarm in yabe, can you please let me know.

khalaniaiyaj, 2023-01-18:
@snehabati Any success on this?

shashankgowdasd, 2023-12-22:
Hi @JoelBender Sir,
May I know any methods available for detecting the alarm services in BACpypes.

JoelBender, 2023-12-22:
Read the **Protocol_Services_Supported** property of the Device Object (Clause 12.11.14) and look at the named bits you get back.

shashankgowdasd, 2023-12-23:
Thanks for the reply sir,
And I have gone through you have suggested me ,but I actually require how to implement the Internsic Alarming for the objects like AnalogInput, AnalogOutput ,BinaryInput and BinaryOutput.

or

Is their any method that support Internsic Alarming for the standard Objects.

And please correct me if my understanding of Alarm Services is incorrect.