JoelBender/bacpypes #354: COV: Need info on cov service functions supported

snehabati, 2020-08-19:
Hi Joel,
I am trying use case using bacpypes.
My use case is that I want to subscribe to the device(bacpypes instance) and if the present value of any cov supported object changes my client needs to be notified of the change in value. 
For this when i searched the code i saw- cov.py has a class(ChangeOfValueServices) and methods - add_subscription(self, cov) and cov_notification(self, cov, request)
I need help as below to continue:
1) what we need to pass in cov param in the functions, 
2) what changes i need to make in add_subscription() function so that I subscribe to device all objects present value

Thanks,
Sneha

JoelBender, 2020-08-19:
The **COVServer.py** sample application (in the samples directory) is the server side which you can test using the **COVClient.py**.  Note that to run the client and server on the same machine at the same time you'll need two settings files with different _address_ values, like "192.168.1.12/24:47808" for the server and "192.168.1.12/24:47809" for the client.

snehabati, 2020-08-24:
Hi Joel,
I am trying to execute the sample COV client server, in client side i want to subscribe to COV. Loos like the args I am giving is not correct can you help me with what kind of args to pass.
(venv) C:\Users\sneha.bati\PycharmProjects\practicepy>python COVClient.py --ini BACPypes_C.ini
> help

Documented commands (type help <topic>):
========================================
EOF    ack      bugin   exit  help    shell
abort  buggers  bugout  gc    reject  subscribe

> help subscribe
subscribe addr proc_id obj_id [ confirmed ] [ lifetime ]
        Generate a SubscribeCOVRequest and wait for the response.

> subscribe 192.168.1.5/24:47809 analogoutput presentValue
ERROR:__main__.SubscribeCOVConsoleCmd:exception: ValueError('invalid format')
Traceback (most recent call last):
  File "C:\Users\sneha.bati\PycharmProjects\practicepy\venv\lib\site-packages\bacpypes\primitivedata.py", line 1696, in __init__
    objType, objInstance = arg.split(':')
ValueError: not enough values to unpack (expected 2, got 1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "COVClient.py", line 119, in do_subscribe
    obj_id = ObjectIdentifier(obj_id).value
  File "C:\Users\sneha.bati\PycharmProjects\practicepy\venv\lib\site-packages\bacpypes\primitivedata.py", line 1701, in __init__
    raise ValueError("invalid format")
ValueError: invalid format
>

snehabati, 2020-08-25:
Hi Joel, can you please let me know what is 'proc_id' argument, from where do we need to get this data.

JoelBender, 2020-08-26:
The format of an object identifier is **_type:instance_** such as **analogInput:12**, where **analogInput** is one of the object type enumerations, see the `ObjectType` list in the `primitivedata` module.  You can also use just numbers like "13:14" will be the same as "multiStateInput:14".

The process identifier is any unsigned value that you want to use, you just have to be consistent when you update or cancel the subscription:

> **13.14.1.1.1 Subscriber Process Identifier**
> This parameter, of type Unsigned32, shall convey a numeric "handle" meaningful to the subscriber. This handle shall be used
to match future re-subscriptions and cancellations from the subscriber with the COV context that exists within the COV-
server device and with confirmed or unconfirmed COV notifications to identify the process within the COV-client that
should receive them. The value zero is reserved for unsubscribed COV notifications as described in Clause 13.7.

snehabati, 2020-08-26:
Hi Joel,
Thanks very much for reply and the information, I am now able to subscribe, had one question if my subscribed obj prop value changes where can I see the notification being sent. I subscribed like below in COVClient.py and I have my COVServer.py running.

(venv) C:\Users\sneha.bati\PycharmProjects\practicepy>python COVClient.py --ini BACPypes_C.ini
> subscribe 192.168.1.5/24:47809 67 analogValue:0
>

JoelBender, 2020-09-03:
When you run the server, add a debugging option `--debug __main_.TestAnalogValueThread` and you should see it waking up (every five seconds in your case) and changing the value of the `test_av` point.  You should see new value, which should be the same one that is returned to your client at the time that it subscribes.

Change your subscription request to include note that you want confirmed notifications and the lifetime of your subscription in seconds:

> subscribe 192.168.1.5/24:47809 67 analogValue:1 true 120

The way that you have written your request, without those two parameters, means "unsubscribe"!