JoelBender/bacpypes #357: How can we set analogInput Value in runtime when a bacpypes instance is running

snehabati, 2020-09-22:
Hi Joel,
I am running one instance of bacpypes in PC1 as application and in another: PC2 I am running another instance and trying to read/write to PC1's object's properties.
My use case is like below:
1.	Set Analog Output's Present Value  = 75.0 
2.	If Analog Output Changes to 75: set the Analog Input to 25 (Here is were I am stuck, how can we change the analog Input Value in this case. I am not talking about the write here I have to set the config and when I read analogInput presentValue I should be able to get 25 )
Can you please help.

snehabati, 2020-09-23:
To add on to this what exactly I m looking for is a dynamically changeable property values in device(PC1) like how it gets changed in the real devices .

JoelBender, 2020-09-23:
I'm not sure I understand completely.  I think that you are asking for an Analog Output Object in PC1 (call it `test_aout`) to be "wired" to an Analog Input Object of the same application (call it `test_ain`).  You will then be using PC2 to change the value of `test_aout` using the Write Property service, then using PC2 again to read the value of `test_ain` using the Read Property service to see that it has changed.  Is this correct?

snehabati, 2020-09-24:
what you have told is right, I want to change test_ain values(like a temp sensor value changes). Would like to know how to achieve this.

SQ9MDD, 2020-10-30:
Hi snehabati

I think you can use  this example for your problem:
[https://github.com/JoelBender/bacpypes/blob/master/samples/mini_device.py](https://github.com/JoelBender/bacpypes/blob/master/samples/mini_device.py)

1. On PC1 setup aplication with two objects inside: "test_aout" and "test_ain" 
2. Inside "DoSomething" class set  test.ain.presentValue = test_aout.presentValue (or another fancy code)
3. From PC2 using Yabe or another OWS set test.aout value and see what happend.

I use this example to read data from JSON server of the weather station.
It works very stable