JoelBender/bacpypes #358: Read Property from Bacnet Ethernet Device.

SQ9MDD, 2020-09-27:
I'm asking the BACNET devices:

"request.pduDestination = Address(addr)"

If the address is in a Bacnet IP or Bacnet MS/TP network everything works.
I ask for a device by its IP address or a pair of network/MAC, for example:

**addr: "192.168.1.3"
addr: "50019:5"**

How can I ask for a Bacnet Ethernet device:
**BacnetDevice: 1100
BacNet ethernet network: 10001
BacMet MAC: 00-40-AE-01-32-B0**

The BBMD (192.168.1.3) is a router to this BacnetEthernet network

I'm lost and I can't construct a correct address in query

Thanks for advice

ChristianTremblay, 2020-09-27:
Try a Whois. If the device can be seen, you will get its address in the
format you need to use.

JoelBender, 2020-09-27:
There are two forms of Ethernet addresses, the one you would usually see if the device is a local station (you and the server are on the same BACnet Ethernet network):
```
>>> from bacpypes.pdu import Address
>>> Address("01:02:03:04:05:06")
<Address 0x010203040506>
```
But that's a mess when you try and add the network number in front:
```
>>> Address("1001:01:02:03:04:05:06")
ValueError: unrecognized format
```
So there an alternative "hex string" format that works for any MAC address length:
```
>>> Address("0x010203040506")
<Address 0x010203040506>
>>> Address("1001:0x010203040506")
<Address 1001:0x010203040506>
```
If you happen to be really an IBM mainframe applications programmer from the 70's you might recognize this, which also shows up in the standard every once in a while:
```
>>> Address("X'010203040506'")
<Address 0x010203040506>
>>> Address("1001:X'010203040506'")
<Address 1001:0x010203040506>
```

SQ9MDD, 2020-09-27:
You are great guys!!!

i am trying: 10001:0x0040ae01329c
wich is 10001:00:40:AE:01:32:9C

it's working!
('10001:0x0040ae01329c', 'analogValue:8', 'presentValue') 15.0

Many thanks again!

closed 2020-09-27 by SQ9MDD