JoelBender/bacpypes #359: EventEnable & Reliability properties are not recognized by YABE

JimitD, 2020-10-07:
Hi Joel, 

I am using COVServer to create BACnet device with AnalogInputObject. I am trying to set two following properties which are unable to recognized by YABE explorer. 

1. eventEnable
2. Reliability

I have set ackedTransitions & notifyType also,. 

Can you please help if I am doing something wrong?

`root_obj_name = AnalogInputObject(
            objectIdentifier=('analogInput', index),
            objectName=obj_name,
            presentValue=0.0,
            covIncrement=0.5,
            eventDetectionEnable=True,
            eventEnable=[1,1,1],
            ackedTransitions=[1,1,1],
            notifyType=1,
            reliability=1,
            outOfService=False,
            eventState=0,
            statusFlags=[0, 0, 0, 0],
            units=19,
        )`

JoelBender, 2020-10-07:
I don't see anything wrong, the clone I made just to see if I can reproduce the problem seems to work fine, and confirmed it with Wireshark, but not YABE.

JimitD, 2020-10-07:
Hello Joel,

Thank you for reply. 

I think YABE explorer is not able recognized EventState & Reliability which is created from BACpypes.  (YABE is able to recognize this properties from inbuild Room Control Simulator) 

Another explorer like InneaBACnet  is able to show all the properties listed in above object creation. 

CAS BACnet explorer is also showing StatusFlags, EventState & OutOfService but other two (EventState & Reliability).

JoelBender, 2020-10-07:
Since I cannot reproduce this with my setup, please capture the traffic between these combinations of explorers and devices and devices then I can see what is different about the way the values are being presented on the wire.  I'm tagging this with "help wanted" for other folks to try the same thing.

JimitD, 2020-10-08:
Hi Joel, 

Looking into Wireshark, some of the explorer does not REQ for most of the properties and hence they are unable to show up in their UI. Thanks for "help wanted" tag.

closed 2020-10-08 by JimitD